espressif/ESP8266_RTOS_SDK #507: ESP8266 IDF编译的BIN生产烧录问题

invention88, 2019-03-11:
ESP8266 IDF SDK开发好BIN文件后，在windows下用“ESPFlashDownloadTool_v3.6.2.exe”工具烧录怎么添加文件和设置地址？

invention88, 2019-03-11:
ESP8266 IDF SDK make后，可以直接生成从地址0烧录的一个target bin文件吗？

FayeY, 2019-03-12:
1. make flash 可以直接烧录，无需 ESPFlashDownloadTool；
2. make 编译完成后会显示烧录的地址信息，可以使用 ESPFlashDownloadTool 按照信息烧录。

invention88, 2019-03-12:
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000 )...
esptool.py v2.4.0
Connecting........_
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:d9:8c:f3
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 8192 bytes to 5631...
Wrote 8192 bytes (5631 compressed) at 0x00000000 in 0.5 seconds (effective 130.5 kbit/s)...
Hash of data verified.
Compressed 481488 bytes to 317215...
Wrote 481488 bytes (317215 compressed) at 0x00010000 in 28.0 seconds (effective 137.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 83...
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.0 seconds (effective 1748.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

从make flash可以看到
0x00010000 位置烧录bin文件。
0x00008000 位置是烧录partitions_singleapp.bin文件吗？（从文件大小看是这个）
0x00000000 位置是烧录boot吗？但大小为8K，而boot文件只有4K。

invention88, 2019-03-12:
知道了：
0x00000000 ：烧录build/bootloader/bootloader.bin
0x00008000 ：位置烧录partitions_singleapp.bin
0x00010000 ：烧录自己的软件

closed 2019-03-12 by invention88