espressif/ESP8266_RTOS_SDK #509: 系统开机异常register map is 3ffffe8c error－有没有人遇到

invention88, 2019-03-12:
2nd stage bootloader[0m
[0;32mI (193) boot: compile time 16:12:54[0m
[0;32mI (194) boot: SPI Speed      : 40MHz[0m
[0;32mI (198) boot: SPI Mode       : QIO[0m
[0;32mI (202) boot: SPI Flash Size : 2MB[0m
[0;32mI (206) boot: Partition Table:[0m
[0;32mI (210) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (217) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (225) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (232) boot:  2 factory          factory app      00 00 00010000 000f0000[0m
[0;32mI (240) boot: End of partition table[0m
[0;32mI (244) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x3e018 (253976) map[0m
[0;32mI (408) esp_image: segment 1: paddr=0x0004e030 vaddr=0x3ffe8000 size=0x0053c (  1340) load[0m
[0;32mI (410) esp_image: segment 2: paddr=0x0004e574 vaddr=0x3ffe853c size=0x0011c (   284) load[0m
[0;32mI (416) esp_image: segment 3: paddr=0x0004e698 vaddr=0x40100000 size=0x066f0 ( 26352) load[0m
[0;32mI (440) boot: Loaded app from partition at offset 0x10000[0m

Core 0 was running in ISR context:

register map is 3ffffe8c error

有没有人遇到这类系统开机异常的情况？

本来一直都很正常，在代码加几条语句，编译后再升级，就成这样了。有时候去掉新加的就可以下运行了。（实际上代码都没跑到修改的地方）

donghengqaz, 2019-03-13:
哪个版本，加的什么？

invention88, 2019-03-13:
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"

#include "nvs_flash.h"

#include <netdb.h>
#include <sys/socket.h>

/* The examples use simple WiFi configuration that you can set via
   'make menuconfig'.

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_WIFI_SSID CONFIG_WIFI_SSID
#define EXAMPLE_WIFI_PASS CONFIG_WIFI_PASSWORD

/* FreeRTOS event group to signal when we are connected & ready to make a request */
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
static const int CONNECTED_BIT = BIT0;

static const char *TAG = "IOT";

/*
static const char *REQUESTDATE = "GET " "http://www.beijing-time.org/time.asp" " HTTP/1.1\r\n"
	"Host: beijing-time.org\r\n"
	"User-Agent: esp-idf/1.0 esp32\r\n"
	"\r\n";
*/
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        /* This is a workaround as ESP32 WiFi libs don't currently
           auto-reassociate. */
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

static void initialise_wifi(void)
[... 250 lines cut ...]
 * Parameters   : none
 * Returns      : none
*******************************************************************************/
void app_main(void)
{
    printf("SDK version:%s\n", esp_get_idf_version());
	
	ESP_ERROR_CHECK( nvs_flash_init() );
	
    initialise_wifi();

	TestFunc();
	
	
    //HTTP_ThreadStart();

	#if (CONFIG_CONSOLE_UART_NUM == 0)
	//Sensor_ThreadStart();
	#endif
	
	//MQTT_ThreadStart();

	//AT_ThreadStart();

}

invention88, 2019-03-13:
用的master版本d847d847e
上面的异常地方是StrToHex函数注释的地方，非常奇怪。
经常会时不时的遇到类似这样的问题，重新整理调整一下函数就好了，如果GetStrDivision函数的调用地方全部删掉，也能正常开机。
感觉像是系统资源问题，但是这些函数需要的资源又是非常小。

invention88, 2019-03-13:
补充一下编译配置文件内容：
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#
CONFIG_IDF_TARGET_ESP32=
CONFIG_IDF_TARGET_ESP8266=y

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-lx106-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_BOOTLOADER_INIT_SPI_FLASH=
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_CHECK_APP_SUM=y
CONFIG_BOOTLOADER_CHECK_APP_HASH=
CONFIG_BOOTLOADER_APP_TEST=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=y
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_SPI_FLASH_MODE=0x0
CONFIG_ESPTOOLPY_FLASHFREQ_80M=
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
CONFIG_SPI_FLASH_FREQ=0x0
CONFIG_ESPTOOLPY_FLASHSIZE_1MB=
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE="2MB"
CONFIG_SPI_FLASH_SIZE=0x200000
CONFIG_ESPTOOLPY_BEFORE_RESET=y
CONFIG_ESPTOOLPY_BEFORE_NORESET=
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=y
CONFIG_ESPTOOLPY_AFTER_NORESET=
CONFIG_ESPTOOLPY_AFTER="hard_reset"
CONFIG_MONITOR_BAUD_9600B=
CONFIG_MONITOR_BAUD_57600B=
CONFIG_MONITOR_BAUD_74880B=y
[... 293 lines cut ...]
#
CONFIG_SSL_USING_MBEDTLS=
CONFIG_SSL_USING_AXTLS=
CONFIG_SSL_USING_WOLFSSL=y

#
# wolfSSL
#
CONFIG_WOLFSSL_DEBUG=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_ADAPTER_GLOBAL_DATA_LINK_IRAM=y

#
# Unified Provisioning
#
CONFIG_ENABLE_UNIFIED_PROVISIONING=

#
# wpa_supplicant
#
CONFIG_LTM_FAST=y

donghengqaz, 2019-05-06:
这个应该是编译的时候产生的非4字节对齐的数据，在 word 级别的寻址访问的时候产生额问题，现在应该已经修复了。

FayeY, 2019-05-20:
由于长时间无更新，此问题暂时关闭。如果使用最新版本仍然有问题，欢迎 reopen issue 继续展开讨论。

closed 2019-05-20 by FayeY