espressif/ESP8266_RTOS_SDK #514: MQTT SSL bug

ochrin, 2019-03-14:
## Environment

- Development Kit:
- SDK version (``git rev-parse --short HEAD`` to get the commit id.): **release/v2.x.x**
- Development Env:      [**Eclipse**]
- Operating System:     [**Ubuntu**]
- Power Supply:         [**USB**]

## Problem Description
MQTT SSL buffer overflow (in readPacket() function) when a publish message is sent "at the same time" as a a PINGREG 

//Detailed problem description goes here.
I am publishing a message every 120s. I set keepalive interval to 60s.
The first message is sent without problem because it is not yet synchronized with PINGREQ
The second message publishing fails due to an overflow buffer in readPacket().

After debug, what I see is this.
1- message is published and PINGREG too almost simultaneously
2- It waits first for PUBREC
3- But it receives PINGREQ first
4- So esp_ssl_read() finishes with n->read_count = 2
5- It continues to waits for PUBREC
6- readPacket calls esp_ssl_read() where n->read_count = 2
7- esp_ssl_read() as n->read_count=2 reads nothing (just move buffer), resets n->read_count and returns
8- readPacket calls decodePacket() to get the lenght of the packet
9- decodePacket() calls esp_ssl_read().
10- esp_ssl_read() as n->read_count = 0 reads two bytes (packet type and packet lg) instead of reading the length.
11- decodePacket returns 0x80 (PUBREC value) as length.
12- readPacket 'crashes' as this length (0x80) is too big for my readbuf.

### Expected Behavior
No buffer overflow

### Actual Behavior
buffer overflow

### Steps to reproduce
Publish a message at the same time a PINGREQ is sent

### Code to reproduce this issue
None

## Debug Logs
None

## Other items if possible
None

ochrin, 2019-03-14:
I fix the problem by adding these 2 lines 
```
		// If length is zero, next read won't be a data read
		if (*buffer == 0)
			n->read_count = 0;
```
in function esp_ssl_read()
```		}
	} else if (n->read_count == 1) { /* read same mqtt packet for the second time */
		if (xTaskCheckForTimeOut(&xTimeOut, &xTicksToWait) == pdTRUE) {
			n->read_count = 0;
			read_buffer[0] = 0;
			return 0;
		}
		os_printk(KERN_DBG "Read2\n");
		n->read_count++;
		*buffer = *(read_buffer + 1);
		// If length is zero, next read won't be a data read
		if (*buffer == 0)
			n->read_count = 0;

		return 1;
	} else if (n->read_count == 2) { /* read same mqtt packet for the third time */
		n->read_count = 0;
```
But not sure my fix as not side effect.

ochrin, 2019-03-14:
By the way, I also don't understand how the original code can work when the length of the packet is coded on more than one byte as esp_ssl_read seems to assume the length of the packet is always coded on one byte:
```
			} else if (FD_ISSET(n->my_socket, &readset)) {
				read_buffer = buffer;
				len = 2; /* len: msg_type(1 octet) + msg_len(1 octet) */

				do {
					rc = SSL_read(n->ssl, read_buffer, len);

					if (rc > 0) {
						recvLen += rc;
					} else if (rc < 0) {
						recvLen = rc;
						return recvLen;
					}
				} while (recvLen < len && xTaskCheckForTimeOut(&xTimeOut, &xTicksToWait) == pdFALSE);

				recvLen = 0;
				len = *(read_buffer + 1); /* len: remaining msg */
				os_printk(KERN_DBG "Read1 2 bytes %x %d\n", read_buffer[0], read_buffer[1]);
				if (len > 0) {
					do {
						rc = SSL_read(n->ssl, read_buffer + 2, len);

```

ustccw, 2019-03-29:
@ochrin @sideralis thanks for your advice and PR.

MQTT based on SSL currently is not provided by our team, it would be our next develep plan. 

thanks.

ochrin, 2019-03-29:
Ok, thanks for the answer.

closed 2019-03-29 by ochrin