espressif/ESP8266_RTOS_SDK #516: "SOCK_RAW" redefined,能否统一下

1XiaSheng, 2019-03-17:
F:/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/include/lwip/sockets.h:165:0: warning: "SOCK_RAW" redefined
 #define SOCK_RAW        3
 ^
In file included from F:/esp/ESP8266_RTOS_SDK/components/esp8266/include/esp_socket.h:23:0,
                 from F:/esp/ht_wifi_new/components/hdw/include/inc_hal.h:32,
                 from F:/esp/ht_wifi_new/components/hdw/src/hdw_debug.c:2:
F:/esp/ESP8266_RTOS_SDK/components/esp8266/include/net/if_socket.h:41:0: note: this is the location of the previous definition
 #define SOCK_RAW 0

1XiaSheng, 2019-03-17:
编译输出的log文件这个警告太多了.

donghengqaz, 2019-03-18:
你为什么要包含 `if_socket.h` 这个头文件，这个是不是给外部使用的，至于 `esp_socket.h` 这个组件，我计划在最新的版本里面移除掉。

FayeY, 2019-03-28:
这个问题是否已经得到解答？

closed 2019-04-01 by FayeY