espressif/ESP8266_RTOS_SDK #517: I2C无法正常使用

rootming, 2019-03-17:
## Environment

- Development Kit:
- IDF version : release/v3.1
- Development Env:      VS code
- Operating System:     MacOS
- Power Supply:         USB

## Problem Description

1. 我的板子是NODEMCU的，使用了GPIO5作为SDA，GPIO4作为SCL来驱动DHT12
2. 使用Arduino的代码试了一下是可以用的(没加外部上拉, 100KHZ)，但是使用3.1的SDK带的I2C驱动无法驱动
3. 加了上拉电阻也没有用
4. 用逻辑分析仪看了一下时序似乎不太对
5. 用示波器看了上升沿下降沿时间是比较正常的
6.偶尔能够正常读取一下，几率很低，同时那个时候用逻辑分析仪看频率只有17.5KHZ

### Code to reproduce this issue

```c

       // 初始化
      int Ret = -1;
      i2c_config_t I2cConfig = {
		.mode = I2C_MODE_MASTER,
		.sda_io_num = DHT12_SDA_IO_NUM,
		.sda_pullup_en = GPIO_PULLUP_ENABLE,
		.scl_io_num = DHT12_SCL_IO_NUM,
		.scl_pullup_en = GPIO_PULLUP_ENABLE
	};

	Ret =  i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER);
	if(Ret != ESP_OK) {
		ESP_LOGE(MOD_TAG, "I2C driver install failed");
		return -1;
	}
	Ret = i2c_param_config(I2C_NUM_0, &I2cConfig);

	if(Ret != ESP_OK) {
		ESP_LOGE(MOD_TAG, "I2C config failed");
		return -1;
	}

        // 读取
        int Ret = -1;
	memset(Buffer, 0, 10);
	i2c_cmd_handle_t I2cHandle = i2c_cmd_link_create();
	i2c_master_start(I2cHandle);
	i2c_master_write_byte(I2cHandle, (u8)0xB8, true);
	i2c_master_write_byte(I2cHandle, (u8)0x0, true);

	i2c_master_start(I2cHandle);
	i2c_master_write_byte(I2cHandle, (u8)0xB9, false);
	i2c_master_read(I2cHandle, Buffer, 5, I2C_MASTER_LAST_NACK);
	i2c_master_stop(I2cHandle);
	Ret = i2c_master_cmd_begin(I2C_NUM_0, I2cHandle, 1000 / portTICK_RATE_MS);
	i2c_cmd_link_delete(I2cHandle);
	return Ret;
```

huozhe0930, 2019-03-17:
试下在任务中循环操作，我这么做的，还算正常，偶尔发现任务开始时段不正常，不清楚原因。我的I2C线太长了，2m。也许是I2C总线干扰问题，串两个电阻在DSA和SCL试试。

rootming, 2019-03-18:
> 试下在任务中循环操作，我这么做的，还算正常，偶尔发现任务开始时段不正常，不清楚原因。我的I2C线太长了，2m。也许是I2C总线干扰问题，串两个电阻在DSA和SCL试试。

好的，谢谢，我试一下

rootming, 2019-03-18:
> 试下在任务中循环操作，我这么做的，还算正常，偶尔发现任务开始时段不正常，不清楚原因。我的I2C线太长了，2m。也许是I2C总线干扰问题，串两个电阻在DSA和SCL试试。

其实我就是在任务中循环操作的，但是是不行，我的线材是20cm差不多，然后Arduino的代码是可以用的。所以比较怀疑是SDK的bug。

xiongyumail, 2019-03-18:
@rootming 能否使用example 中的io2,io14 测试一下。

rootming, 2019-03-18:
> @rootming 能否使用example 中的io2,io14 测试一下。

好的，我会试一下，不过nodemcu的板子GPIO2被LED占用，不知道是否会有影响

rootming, 2019-03-18:
### 补充一下逻辑分析仪的图片

## Arduino的代码
[image]

## SDK的代码
[image]

huozhe0930, 2019-03-18:
乐鑫的I2C封装的还是挺好用的，我用了2个I2c端口，都能正常工作。你就只操作I2c,其他操作先停掉呢？试着换下端口，我用了2\4\5\14这几个端口。
//No. 0 i2c 
#define I2C_MASTER_0_SCL_IO          2                /*!< gpio number for I2C master clock */
#define I2C_MASTER_0_SDA_IO          14               /*!< gpio number for I2C master data  */
#define I2C_MASTER_NUM_0             I2C_NUM_0        /*!< I2C port number for master dev */
//No. 1 i2c for other purpose
#define I2C_MASTER_1_SCL_IO          5                /*!< gpio number for I2C master clock */
#define I2C_MASTER_1_SDA_IO          4                /*!< gpio number for I2C master data  */
#define I2C_MASTER_NUM_1             I2C_NUM_1        /*!< I2C port number for master dev */

#define I2C_MASTER_TX_BUF_DISABLE    0                /*!< I2C master do not need buffer */
#define I2C_MASTER_RX_BUF_DISABLE    0                /*!< I2C master do not need buffer */

i2c_master_cmd_begin_static() ---这个是i2c的核心函数了，在进入这之前，会关中断（看代码是这样的，通过调用ENTER_CRITICAL()，里面有一步是关中断）。是不是中断惹的祸呢？

rootming, 2019-03-18:
> 乐鑫的I2C封装的还是挺好用的，我用了2个I2c端口，都能正常工作。你就只操作I2c,其他操作先停掉呢？试着换下端口，我用了2\4\5\14这几个端口。
> //No. 0 i2c
> #define I2C_MASTER_0_SCL_IO 2 /*!< gpio number for I2C master clock _/ #define I2C_MASTER_0_SDA_IO 14 /_!< [...]

试过关闭其他操作，不过没有啥用。他的实现我也看过了。。。233

rootming, 2019-03-18:
> @rootming 能否使用example 中的io2,io14 测试一下。

试了，不行的，在SCL和SDA之间加电阻也不行

huozhe0930, 2019-03-18:
scl串接100欧姆以下电阻，sda串接100欧姆以下电阻；不是SCL和SDA之间加电阻。I2C总线的总电容是多少？电容太大，可以尝试减少上拉电阻阻值。

rootming, 2019-03-18:
> scl串接100欧姆以下电阻，sda串接100欧姆以下电阻；不是SCL和SDA之间加电阻。I2C总线的总电容是多少？电容太大，可以尝试减少上拉电阻阻值。

好的我试一下。我只是比较好奇ESP8266用Arduino的实现是可以直接使用的，上拉电阻也不用加，直接用内部的就行，使用SDK后就不行了。

huozhe0930, 2019-03-18:
我看了你发的图形，感觉SDK，SCL在总线空闲时的电平有点儿不对呢，应该是高电平啊，你的是低电平呢？

rootming, 2019-03-18:
> 我看了你发的图形，感觉SDK，SCL在总线空闲时的电平有点儿不对呢，应该是高电平啊，你的是低电平呢？

[image]
看代码应该是没收到ACK

huozhe0930, 2019-03-18:
> 
> 
> > 我看了你发的图形，感觉SDK，SCL在总线空闲时的电平有点儿不对呢，应该是高电平啊，你的是低电平呢？
> 
> [image]
> 看代码应该是没收到ACK，所以就维持低电平了吧

看实现函数里的延时，SDK 3.1版本的i2c，速率应该在250k左右。改代码，加延时试试呢。SCL从0->1后，几乎没有延时，SCL变成1后，也只延时了1us。
按你描述的，NODEMCU的I2C才100K的速率，明显比SDK的要低。要不增加延时试试。

rootming, 2019-03-18:
> > > 我看了你发的图形，感觉SDK，SCL在总线空闲时的电平有点儿不对呢，应该是高电平啊，你的是低电平呢？
> > 
> > 
> > [image]
> > 看代码应该是没收到ACK，所以就维持低电平了吧
> 
> 看实现函数里的延时，SDK 3.1版本的i2c，速率应该在250k左右。改代码，加延时试试呢。SCL从0->1后，几乎没有延时，SCL变成1后，也只延时了1us，就开始读SDA了。
> 按你描述的，NODEMCU的I2C才100K的速率，明显比SDK的要低。要不增加延时试试。

3.1 SDK我逻辑分析仪看是100khz, 代码里延迟1us实际并不是延迟1us, 会不会是主频的原因？ 虽然我并没有改过主频。

xiongyumail, 2019-03-29:
@rootming 请问这个问题有什么进展了么？

rootming, 2019-03-29:
> @rootming 请问这个问题有什么进展了么？

后面我换了一种总线驱动传感器，I2C没有继续尝试，传感器是DHT12

FayeY, 2019-04-01:
这个问题先行关闭，如果后续有其他更新，欢迎 reopen。

closed 2019-04-01 by FayeY