espressif/ESP8266_RTOS_SDK #519: 使用mqtt 例子收到subscribe 自动断开会话

panguoxian, 2019-03-19:
如果MQTT_RECV_CYCLE时间设置的太小，比如100ms。收到subscribe 会自动关闭会话。
打印如下：
W (25146) mc: mqtt close session
E (25146) mc: MQTTRun cycle failed

当我把MQTT_RECV_CYCLE配置为500ms，如果不停发送主题，还是会自动关闭mqtt连接。

不是很理解例子里面配置的
MQTT_SEND_CYCLE、MQTT_RECV_CYCLE、MQTT_PUBLISH_INTERVAL
有朋友可以解释一下吗？

个人理解：
MQTT_SEND_CYCLE和MQTT_RECV_CYCLE 是接收和发送的间隔时间，即多长时间接收或发送一次
MQTT_PUBLISH_INTERVAL是多么时间publish?

ustccw, 2019-03-29:
@panguoxian 你是有什么特别需求吗?  `MQTT_RECV_CYCLE` 默认是 0, 通常用户不修改.

`MQTT_SEND_CYCLE` 表示发送一条 message, 最大不能超过的时间限制 cycle.
`MQTT_RECV_CYCLE` 表示去接收 MQTT 包间隔.
`MQTT_PUBLISH_INTERVAL` 表示设备发布 MQTT 消息间隔.

具体的在 `make menuconfig` 配置项里 HELP 都有说明的.

ustccw, 2019-03-29:
@panguoxian 
[image]

panguoxian, 2019-04-01:
@ustccw 非常感谢。！！！

closed 2019-04-17 by FayeY