espressif/ESP8266_RTOS_SDK #520: esp_http_client支持https的问题

MinjunHuang, 2019-03-19:
## Environment

- Development Kit :  ESP8266 SDK version:v3.2-dev-162-g8378b1a
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB
-Web Server：         nginx + tornado 4.4.2

## Problem Description
1. 我的应用采用http协议向服务器上报数据，使用esp_http_client发送http post请求，采用HTTP时工作正常，服务器能够获取到数据; 采用HTTPS时候，服务器上https请求中获得的body是乱码。
2. 我怀疑是不是服务器的问题，所以我使用esp_tls_conn_xxx函数向服务器发送https post请求，服务器也能正常的获取数据。
3. 因此我猜测可能问题在esp_http_client，或者是esp_http_client在http和https情况下的使用有没有什么差异？
我的代码片段：
#ifdef SUPPORT_HTTPS    
    esp_http_client_config_t config = {
        .url = "https://www.baisu.xyz/post",
        .event_handler = _http_event_handler,
        .timeout_ms = 5000,
        .cert_pem = server_root_cert_pem_start,
    };
#else
    esp_http_client_config_t config = {
        .url = "http://www.baisu.xyz/post",
        .event_handler = _http_event_handler,
        .timeout_ms = 3000,
    };
#endif

    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_err_t err;
    
     esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_header(client, "Content-Type", "application/x-www-form-urlencoded;charset=utf-8");
    esp_http_client_set_header(client, "Accept", "text/plain,application/json");
      esp_http_client_set_post_field(client, data, strlen((char*)data));
    while (1) {
        err = esp_http_client_perform(client);
        if (err != ESP_ERR_HTTP_EAGAIN) {
            break;
        }
    }

xuhongv, 2019-03-26:
- https需要本地在服务器的证书

jitin17, 2019-03-26:
@MinjunHuang Can you call `esp_http_client_set_header(client, "Content-Type", "application/x-www-form-urlencoded;charset=utf-8");` after `esp_http_client_set_post_field(client, data, strlen((char*)data));` and see if it works?

MinjunHuang, 2019-03-28:
@jitin17 
  it doesn't work, i got the same result as before, i'll check whether something i can do to get the correct data at server side.     

@xuhongv 
   now that i got right data  using esp_tls_conn_xxx functions, i guess the cert file work.
 
anyway, thanks for your suggestion.

closed 2019-03-28 by FayeY