espressif/ESP8266_RTOS_SDK #523: esp8266 make monitor显示ets Jan  8 2013,rst cause:2, boot mode:(1,6)之后没任何输出

yqzzj, 2019-03-22:
运行ESP8266_RTOS_SDK/examples/get-started/下例子，编译烧写成功后，make monitor没有输出log，如下：
$ make monitor
MONITOR
--- idf_monitor on /dev/ttyUSB0 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
之后便没log了，更改晶振设置为另外三个值、更改波特率为115200均无变化，是程序没有自动运行吗？

donghengqaz, 2019-03-22:
试试 make flash monitor，然后手动复位重启板子。

yqzzj, 2019-03-22:
一样的 
...
Wrote 10816 bytes (7181 compressed) at 0x00000000 in 0.6 seconds (effective 135.5 kbit/s)...
Hash of data verified.
Compressed 220864 bytes to 152448...
Wrote 220864 bytes (152448 compressed) at 0x00010000 in 13.4 seconds (effective 131.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 83...
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.0 seconds (effective 1397.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB0 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

之后就没了

27-301, 2019-03-22:
你是什么板子？

8266 的 boot 有两种模式，flash boot & uart boot，通过 gpio 0 控制，当上电时 gpio0 为低时是 uart boot，为高时是 flash boot。

具体模式可以通过上电的 log 查看：
```
ets Jan 8 2013,rst cause:2, boot mode:(1,6)
```
这里 boot mode:(`1`,6)，1 代表的是 uart boot 模式，你烧写完，把 gpio0 拉高，上电。上电打印 boot mode:(`3`,6)，就能 boot 起来了。

yqzzj, 2019-03-22:
谢谢  可以了

closed 2019-03-22 by yqzzj