espressif/ESP8266_RTOS_SDK #525: Task stack [IDLE] stack ? why?

1XiaSheng, 2019-03-25:
***ERROR*** A stack overflow in task IDLE has been detected.
E (165007) ABORT: Error found and abort!

Task stack [IDLE] stack from [0x3fff08d8] to [0x3fff0bd4], total [768] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c 

3fff0b00  0x4026967c 0xfe7fffff 0x600005e8 0x4022d41a 0x00000000 0x0000e48e 0x00000020 0x4010049c 0x00000000 0x00000000 0x00000000 0x4010056d 0x09d3e9a9 0x3ffe9cc8 0x3ffe9c74 0x401051d4 
3fff0b40  0x40104f90 0x000c3500 0x12435f67 0x40273ce8 0x40104fb8 0x402745f4 0x00000030 0x402745f4 0x3fff0ba0 0x00000000 0x00000000 0x60000624 0x0000df51 0x00003a36 0x60009830 0x4026967c 
3fff0b80  0x000000eb 0x0000001e 0x000001e0 0x3ffe8004 0x00000000 0x00000000 0x3ffe8004 0x00000004 0x00000000 0x00000000 0x00000000 0x4022d428 0x00000000 0x00000000 0x00000000 0x4022c0ec 
3fff0bc0  0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 

        PC: 0x40248cd6        PS: 0x00000031        A0: 0x40248cd4        A1: 0x3fff0b00
        A2: 0x00000000        A3: 0x00000000        A4: 0x00000001        A5: 0x00000029
        A6: 0x40248558        A7: 0x00000000        A8: 0x40107d38        A9: 0x00000090
       A10: 0x402123e6       A11: 0x20646e61       A12: 0x00000000       A13: 0x00000000
       A14: 0x00000000       A15: 0x00000000       SAR: 0x00000004  EXCCAUSE: 0x0000001d

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset

1XiaSheng, 2019-03-26:
又碰到了,
[image]

1XiaSheng, 2019-03-28:
[image]

1XiaSheng, 2019-03-28:
register map is 3ff7ffd0 error

1XiaSheng, 2019-03-28:
有没有什么方法找到任务溢出报出的 寄存器 关联位置 register map is 3ff7ffd0  ,3ff7ffc0 ,

donghengqaz, 2019-03-28:
你会在 idle 回调函数里面做复杂的业务？

1XiaSheng, 2019-03-28:
不会啊，我的任务创建之后就是干自己的活， 不会影响到IDLE。这个内存溢出最难查了，最近项目使用更新后的SDK编译就是经常会死掉

donghengqaz, 2019-04-17:
最新的更新里面已经修复了这个问题。

closed 2019-04-24 by FayeY