espressif/ESP8266_RTOS_SDK #526: Using RTC Memory to store/load data after reset

Bennji94, 2019-03-25:
I want to store data in the rtc memory and load it after a reset. Is there a function to store and load data from rtc memory?  Using a variable with RTC_DATA_ATTR didn't worked.

donghengqaz, 2019-03-28:
You may use master branch to work it.

Bennji94, 2019-03-29:
I used the master branch but in the following example the rtc_mem keeps  0 after each reset.

```cpp
#include <stdio.h>

#include "esp_system.h"
#include "esp_log.h"
#include "esp_attr.h"

RTC_DATA_ATTR int rtc_mem;
static const char* TAG = "main";

/******************************************************************************
 * FunctionName : app_main
 * Description  : entry of user application, init user function here
 * Parameters   : none
 * Returns      : none
*******************************************************************************/
void app_main(void)
{
   	printf("SDK version:%s\n", esp_get_idf_version());
	if (rtc_mem != 0) {
			ESP_LOGI(TAG, "rtc_mem!=0");
			rtc_mem = 0;
		} else {
			ESP_LOGI(TAG, "rtc_mem==0");
			rtc_mem = 1;
		}
}
```

Logs
```
I (53) boot: ESP-IDF v3.2-dev-189-g0787de1-dirty 2nd stage bootloader
I (53) boot: compile time 04:04:25
I (54) qio_mode: Enabling default flash chip QIO
I (62) boot: SPI Speed      : 40MHz
I (68) boot: SPI Mode       : QIO
I (75) boot: SPI Flash Size : 2MB
I (81) boot: Partition Table:
I (86) boot: ## Label            Usage          Type ST Offset   Length
I (98) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (109) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (121) boot:  2 factory          factory app      00 00 00010000 000f0000
I (132) boot: End of partition table
I (139) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x2c16c (180588) map
I (212) esp_image: segment 1: paddr=0x0003c184 vaddr=0x60001200 size=0x00004 (     4) load
I (213) esp_image: segment 2: paddr=0x0003c190 vaddr=0x3ffe8000 size=0x0050c (  1292) load
I (224) esp_image: segment 3: paddr=0x0003c6a4 vaddr=0x3ffe850c size=0x0011c (   284) load
I (238) esp_image: segment 4: paddr=0x0003c7c8 vaddr=0x40100000 size=0x05758 ( 22360) load
0x40100000: _stext at ??:?

I (258) boot: Loaded app from partition at offset 0x10000
I (304) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (314) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (484) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
SDK version:v3.2-dev-189-g0787de1-dirty
I (494) main: rtc_mem==0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
0x40100000: _stext at ??:?

tail 12
chksum 0x97
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x1e
ho 0 tail 12 room 4
load 0x3ffe8420, len 3552, room 12 
tail 4
chksum 0x41
I (53) boot: ESP-IDF v3.2-dev-189-g0787de1-dirty 2nd stage bootloader
I (53) boot: compile time 04:04:25
I (54) qio_mode: Enabling default flash chip QIO
I (62) boot: SPI Speed      : 40MHz
I (68) boot: SPI Mode       : QIO
I (75) boot: SPI Flash Size : 2MB
I (81) boot: Partition Table:
I (86) boot: ## Label            Usage          Type ST Offset   Length
I (98) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (109) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (121) boot:  2 factory          factory app      00 00 00010000 000f0000
I (132) boot: End of partition table
I (139) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x2c16c (180588) map
I (212) esp_image: segment 1: paddr=0x0003c184 vaddr=0x60001200 size=0x00004 (     4) load
I (213) esp_image: segment 2: paddr=0x0003c190 vaddr=0x3ffe8000 size=0x0050c (  1292) load
I (224) esp_image: segment 3: paddr=0x0003c6a4 vaddr=0x3ffe850c size=0x0011c (   284) load
I (238) esp_image: segment 4: paddr=0x0003c7c8 vaddr=0x40100000 size=0x05758 ( 22360) load
0x40100000: _stext at ??:?

I (258) boot: Loaded app from partition at offset 0x10000
I (304) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (314) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (484) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
SDK version:v3.2-dev-189-g0787de1-dirty
I (494) main: rtc_mem==0

```

27-301, 2019-04-12:
@Bennji94 rtc memory only be kept after deep sleep wake up.

FayeY, 2019-04-17:
Closing as this question appears to be answered, please feel free to reopen if this is still an issue.
Thanks.

closed 2019-04-17 by FayeY