espressif/ESP8266_RTOS_SDK #528: PWM phase semantics is error-prone

sysprogs, 2019-03-25:
The values of -180 and 180 suggest that the phase is specified in degrees, so +180 would mean "half a period later" and +90 would mean "quarter of a period later". However, currently +90 translates to "half period later" and +45 - to "quarter period later".

Also since the comment in pwm_set_phase() doesn't mention the exact semantics, it does likely confuse near 100% of users trying the API for the first time.

xiongyumail, 2019-03-29:
The repaired version is being merged. Thank you！

closed 2019-04-01 by wujiangang

sysprogs, 2019-04-01:
Thanks, looks great. Just one concern: does that commit change the behavior of the existing API? If yes, it will break everyone's code that relies on the old phase values.

We haven't double-checked, but if it this is the case, please consider renaming the APIs related to phase, so that people would get a compile-time error (preferably with a quick comment explaining what has changed) instead of just a silent change in behavior.