espressif/ESP8266_RTOS_SDK #534: boot文件内容被篡改

tonyecho, 2019-03-28:
最近发现用8266SDK开发的智能家居产品，工作半年左右后8266模块无法链接到指定服务器，用串口工具看上电打印信息，boot文件部分字节被篡改，导致模块启动后无法运行应用代码：
--------------------------------------------------------
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe4
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x85
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd9
csum 0xd9
csum err
ets_main.c

tonyecho, 2019-03-28:
[image]
[image]

donghengqaz, 2019-03-28:
这个是所有产品都出现这个问题，还是就是部分？你们业务逻辑里面会写 flash 吗？

donghengqaz, 2019-03-29:
你们用的是官方提供的模组？可以从 flash 多读取一些数据出来看看，还有没有其他的地方被篡改。

FayeY, 2019-04-01:
想请问，这个问题有更新吗？

tonyecho, 2019-04-01:
> 这个是所有产品都出现这个问题，还是就是部分？你们业务逻辑里面会写 flash 吗？

少部分有问题，业务逻辑有写flash

FayeY, 2019-06-27:
从 flash 多读取一些数据出来，看看有没有其他的地方被篡改? 有可能是 flash 的某些 bit 发生了跳变

closed 2019-07-15 by FayeY