espressif/ESP8266_RTOS_SDK #535: esp_wifi_set_max_tx_power() not reducing transmit power level

gregorHerburger, 2019-03-28:
Hello,

i use the ESP8266 in WIFI_MODE_AP on ESP8266_RTOS_SDK v3.0. I want to reduce the wifi sending power. When i call esp_wifi_set_max_tx_power() with different values, it doesn't seem to change the transmitted power. When calling esp_wifi_set_max_tx_power it always returns 20.

```
    if (ESP_OK != esp_wifi_set_max_tx_power(8))
    {
        ESP_LOGE(TAG, "Failed to reduce wifi tx power");
    }
```
never gives the Error Message. esp_wifi_set_max_tx_power is called after esp_wifi_start();

How can i reduce the tx power in software?
Best regards.

gregorHerburger, 2019-03-29:
Hello,

I observed the problem a bit more. It seems to be there are two bugs:
1. esp_wifi_set_max_tx_power() not working.
2. esp_wifi_get_max_tx_power() always showing 20.

Both bugs seem to be fixed in the current master:
1. Fixed in commit eacf6c3b37da7bb28fd184be03868f0aab0c338f 
2. Fixed in commit 181c3de1b4306a053596850ae21cd33fa72dece8 

Could you make a backport of these two bugfixes to the release v3.0? 
This is really urgent for our product.

Another thing i observed is when, esp_wifi_set_max_tx_power() becomes negative, the tx power is to maximal again. This is other than in the description where it says:
 [-128, 7]: level5 - 14dBm
So negative values seem to lead to max power. 
Is this observation correct?

Best regards.

FayeY, 2019-04-17:
Would you like to upgrade to the new version V3.1? Since the V3.0 is the first version of the IDF style SDK, V3.1 should be more stable than V3.0. It is always suggested to use the new version SDK.
Thanks.

Junhao-Espressif, 2019-04-17:
Hi @gregorHerburger 

We will not update this api on V3.0. If you really need this, 
Please use ```void phy_set_most_tpw( int8_t power); ``` to set tx power.

```
  *            - [82, 127]: level0
  *            - [78,  81]: level1
  *            - [74,  77]: level2
  *            - [68,  73]: level3
  *            - [64,  67]: level4
  *            - [56,  63]: level5
  *            - [49,  55]: level5 - 2dBm
  *            - [33,  48]: level5 - 6dBm
  *            - [25,  32]: level5 - 8dBm
  *            - [13,  24]: level5 - 11dBm
  *            - [ 1,  12]: level5 - 14dBm
  *            - [-128, 0]: level5 - 17.5dBm
```

FayeY, 2019-04-24:
Closing as this question appears to be answered. Please feel free to reopen if this is still an issue. Thanks.

closed 2019-04-24 by FayeY