espressif/ESP8266_RTOS_SDK #536: section `.text2.40' will not fit in region `iram1_0_seg' since Merge branch 'feature/update_libphy' into 'master'

Bennji94, 2019-03-29:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 8c3d835
- Development Env:    Make
- Operating System:    Ubuntu
- Power Supply:         USB

## Problem Description

Since the commit Merge branch 'feature/update_libphy' into 'master' (8c3d835f2daeba5c72aeee348af2e804f2801f4e) I get the error message "section `.text2.40' will not fit in region `iram1_0_seg'  when linking. I didn't use any IRAM_ATTR in my code and already deactivated all iram options in menuconfig. Is there any option to reduce the code size in iram?

donghengqaz, 2019-04-01:
Our link file has changed, and add some other section like this:
```
    *(.rodata2.* .rodata2 .literal2.* .literal2 .text2.* .text2)
```

FayeY, 2019-04-17:
Is this issue solved?

closed 2019-04-24 by FayeY