espressif/ESP8266_RTOS_SDK #537: http server freeze

WallaceWilliam, 2019-03-30:
- Development Kit: 8266 D1 WEMOS
- IDF version ESP8266_RTOS_SDK master
- Development Env:      Platformio
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
Testing was conducted on an example protocols/http_server/simple
after the access point is turned off (event message=SYSTEM_EVENT_STA_DISCONNECTED) httpd is freezed.
This is part of the httpd_stop() function code
```
    memset(&msg, 0, sizeof(msg));
    msg.hc_msg = HTTPD_CTRL_SHUTDOWN;
    cs_send_to_ctrl_sock(hd->msg_fd, hd->config.ctrl_port, &msg, sizeof(msg));

    ESP_LOGD(TAG, LOG_FMT("sent control msg to stop server"));
    **while (hd->hd_td.status != THREAD_STOPPED) {
        httpd_os_thread_sleep(100);
    }**
```
a command is sent to the server to stop and the server is expected to stop

> D (51450) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:line, ssid_len:4, bssid:3b5:e2d:ac, reason:200
> V (51450) event: enter default callback
> D (51460) udp_sync: UDP sync close port 68
> netif_set_ipaddr: netif address being changed
> netif: IP address of interface en set to 0.0.0.0
> netif: netmask of interface en set to 0.0.0.0
> netif: GW address of interface en set to 0.0.0.0
> D (51460) tcpip_adapter: if0 start ip lost tmr: enter
> D (51460) tcpip_adapter: if0 start ip lost tmr: interval=120
> V (51470) event: exit default callback
> I (51470) APP: SYSTEM_EVENT_STA_DISCONNECTED
> lwip_sendto(2, data=0x3fff3050, short_size=12, flags=0x0 to=127.0.0.1 port=32768
> D (51480) udp_sync: UDP sync regitser sock msg 0x3fff2f90
> D (51480) httpd: httpd_stop: sent control msg to stop server
> D (51480) udp_sync: UDP sync ack msg 0x3fff2f90

for Manage in-coming connection or data requests
in _static esp_err_t httpd_server(struct httpd_data *hd)_
call _int active_cnt = select(maxfd + 1, &read_set, NULL, NULL, NULL);_

> lwip_select(2, 0x3fff4510, 0, 0, tvsec=-1 tvusec=-1)
however, since STA_DISCONNECTED and IP = 0.0.0.0 the server will never receive a command to stop

supreetd21, 2019-04-08:
Hi @WallaceWilliam,

Thank you for pointing out the issue and well-documenting the question.

Looks like the UDP control socket did not receive the message sent in the disconnection sequence. The problem here looks like the 'CONFIG_LWIP_NETIF_LOOPBACK' option hasn't been enabled.

This option enables the messages with destination address 127.0.0.1 to loop back to the interface. This LWIP value is set to false by default but should be changed. Will add a fix for this issue.

Meanwhile, you could try to get around the issue by setting the LWIP_NETIF_LOOPBACK in menuconfig. You will find it at COMPONENT_CONFIG-> LWIP-> Enable per-interface loopback.

Do let me know if this solves the issue.

WallaceWilliam, 2019-04-08:
it work!!!

supreetd21, 2019-04-10:
Glad that it worked.
A fix has been added(https://github.com/espressif/ESP8266_RTOS_SDK/commit/2598b7ec216119acfeb7b1de2042e246b419b345) wherein the Loopback options have been set in the sdkconfig.defaults. 
Run _make defconfig_ before _make_ to ensure the options have been applied in the firmware.

closed 2019-04-10 by donghengqaz