espressif/ESP8266_RTOS_SDK #539: Cannot connect to AP

christian-0s, 2019-04-01:
## Environment

- Development Kit: ESP8266_RTOS_SDK
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 0460e534
- Development Env:      Make
- Operating System:     Ubuntu 18.04
- Power Supply:         USB

## Problem Description
ESP8266 cannot connect to my AP (Netgear DG834Gv2) when using WPA TKIP, it works with this AP if network is unprotected. 
It works using another AP (e.g. an Android phone). 
ESP32 and other wifi devices can connect just fine with my Netgear AP.

### Code to reproduce this issue

I'm using the "simple_wifi" example included in SDK.

## Debug Logs

```
2019-04-01 18:21:08.309:  scandone
2019-04-01 18:21:08.309:  state: 0 -> 2 (b0)
2019-04-01 18:21:08.310:  state: 2 -> 3 (0)
2019-04-01 18:21:08.313:  state: 3 -> 0 (1200)
2019-04-01 18:21:08.317:  D (3403) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:*DELETED*, ssid_len:6, bssid:*DELETED*, reason:203
2019-04-01 18:21:08.330:  V (3403) event: enter default callback
2019-04-01 18:21:08.335:  D (3413) tcpip_adapter: if0 start ip lost tmr: enter
2019-04-01 18:21:08.343:  D (3423) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3fff47f8 interval=120 ip=0
2019-04-01 18:21:08.356:  V (3433) event: exit default callback

```
[debug.log](https://github.com/espressif/ESP8266_RTOS_SDK/files/3030477/log.log)

[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/3030465/sdkconfig.txt)

27-301, 2019-04-02:
hi, @christian-0s can you use wireshark to capture the wifi packets?

christian-0s, 2019-04-02:
Hi @27-301 
I noticed that AP was set in g mode only, so I changed setting to b and g, the debug output is slighly different:
```
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (100)
D (2207873) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:Sakura, ssid_len:6, bssid:00:0f:b5:d5:a0:9e, reason:203
V (2207873) event: enter default callback
D (2207873) tcpip_adapter: if0 start ip lost tmr: enter
D (2207883) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3fff47f8 interval=120 ip=0
V (2207903) event: exit default callback
```
In attachment my attempt to capture wifi packets with Wireshark, I hope I did it correctly because I'm not familiar with this tool.

[wireshark.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3035156/wireshark.zip)

Junhao-Espressif, 2019-04-09:
Hi @christian-0s 

Could you connect this AP with your phone and capture the wifi packets when it works well.

state: 3 -> 0 (1200) means Invalid group cipher
state: 3 -> 0 (100) means  Unspecified reason

christian-0s, 2019-04-09:
Hi @Junhao-Espressif 

During previous capture I realized that I had b mode disabled on my AP. So I enabled it and then I got the last failure message "state: 3 -> 0 (100) means Unspecified reason"
Attached a capture of a successful connection between a ESP32 based board and my AP.

Best Regards
[esp32.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3058456/esp32.zip)

Junhao-Espressif, 2019-04-10:
Hi @christian-0s 

Could you try with the library in attachment at master version? 

Thank you!

[libnet80211.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3062346/libnet80211.zip)

christian-0s, 2019-04-10:
Using this library ESP8266 has no problems connecting to my AP! Many thanks!
In attachment the capture of successful connection.
Debug log:
```
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with Sakura, channel 13
D (3267) nvs: nvs_set_blob sta.bssid 6
D (3277) event: SYSTEM_EVENT_STA_CONNECTED, ssid:Sakura, ssid_len:6, bssid:00:0f:b5:d5:a0:9e, channel:13, authmode:1073678720
V (3277) event: enter default callback
D (3287) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (3297) tcpip_adapter: if0 start ip lost tmr: enter
D (3307) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x40106858 interval=120 ip=0
D (3317) udp_sync: UDP sync cache udp 0x40106910
D (3327) udp_sync: UDP sync clear udp 0x40106910
D (3327) tcpip_adapter: dhcp client start successfully
V (3337) event: exit default callback
D (3347) udp_sync: UDP sync cache udp 0x40106910
D (3347) udp_sync: UDP sync clear udp 0x40106910
D (3827) tcpip_adapter: dhcpc time(ms): 0

D (4327) tcpip_adapter: if0 dhcpc cb
D (4327) tcpip_adapter: if0 ip changed=1
D (4327) event: SYSTEM_EVENT_STA_GOT_IP, ip:192.168.0.24, mask:255.255.255.0, gw:192.168.0.1
V (4337) event: enter default callback
I (4337) event: sta ip: 192.168.0.24, mask: 255.255.255.0, gw: 192.168.0.1
V (4347) event: exit default callback
I (4357) simple wifi: got ip:192.168.0.24
D (4357) tcpip_adapter: ip:192.168.0.24,mask:255.255.255.0,gw:192.168.0.1
```

[esp8266_libnet80211_mod.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3066010/esp8266_libnet80211_mod.zip)

closed 2019-04-15 by wujiangang