espressif/ESP8266_RTOS_SDK #541: snprintf bug

nickfox-taterli, 2019-04-04:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):eb6d4f7f
- Development Env:      Ubuntu 18.04 x64
- Operating System:     Ubuntu
- Power Supply:            External 3.3V@1A

## Problem Description

使用newlib(nano)时候不报告错误,但是snprintf不兼容标准snprintf,在%f的情况下出错.如果使用newlib(normal),则出现错误,由于单独测试snprintf下没有出现错误.所以对目前这个错误,不明白是什么原因,但是唯一差异是,使用newlib(nano)不出错,使用newlib(normal)出错,使用newlib(normal)但屏蔽snprintf相关代码,也不出错.

### Expected Behavior

snprintf应正确处理字符串.

### Actual Behavior

只要遇到了vsnprintf代码并且使用newlib(normal)就出错.

### Steps to repropduce

1. 使用我移植的工程https://github.com/nickfox-taterli/esp8266_qcloud/
2. 使用newlib(normal)编译,并在过程中遇到%f.

### Code to reproduce this issue

```cpp
int HAL_Snprintf(_IN_ char *str, const int len, const char *fmt, ...)
{
    va_list args;
    int rc;

    va_start(args, fmt);
    rc = vsnprintf(str, len, fmt, args);
    va_end(args);

    return rc;
}
```
## Debug Logs

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7600, room 16 
0x40100000: _stext at ??:?

tail 0
chksum 0xc2
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x7b
load 0x3ffe8420, len 4280, room 8 
tail 0
chksum 0xba
I (44) boot: ESP-IDF v3.2-dev-185-geb6d4f7f-dirty 2nd stage bootloader
I (44) boot: compile time 15:45:48
I (44) qio_mode: Enabling default flash chip QIO
D (48) boot: bootloader initialize SPI flash clock and I/O
D (56) boot: magic e9
D (59) boot: segments 03
D (63) boot: spi_mode 00
D (66) boot: spi_speed 0f
D (70) boot: spi_size 02
I (73) boot: SPI Speed      : 80MHz
I (78) boot: SPI Mode       : QIO
I (83) boot: SPI Flash Size : 1MB
D (88) boot: mapped partition table 0x8000 at 0x40208000
D (95) flash_parts: partition table verified, 3 entries
[... 130 lines cut ...]
3fff7480  0x0003a980 0x00000101 0x4024388c 0x401065fc 0x00000000 0x00000000 0x40214614 0x40232be6 0xffffffff 0x00000001 0x3fff5640 0x00000000 0x00006568 0xffffffff 0x40211030 0x40211028 
0x4024388c: _shadow_event_handler at /home/taterli/qcloud/esp8266_qcloud/qcloud/qcloud-iot-sdk-embedded-c/src/shadow/src/shadow_client.c:637

0x40232be6: mqtt_loop at /home/taterli/qcloud/esp8266_qcloud/main/main.c:382

3fff74c0  0x4021103c 0x000007d0 0x0003a980 0x00000101 0x402323c4 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40232df9 
0x402323c4: mqtt_event_handler at /home/taterli/qcloud/esp8266_qcloud/main/main.c:204

0x40232df9: mqtt_main at /home/taterli/qcloud/esp8266_qcloud/main/main.c:465

3fff7500  0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 

        PC: 0x4000bf80        PS: 0x00000030        A0: 0x40234e5a        A1: 0x3fff6e80
0x40234e5a: _svfprintf_r at /home/dongheng/workspace/espressif/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1529

        A2: 0x00000000        A3: 0xfffffffc        A4: 0x000000ff        A5: 0x0000ff00
        A6: 0x00ff0000        A7: 0xff000000        A8: 0xffffffff        A9: 0x3fff6e90
       A10: 0x3fff6e90       A11: 0x40213a3b       A12: 0x3fff6e90       A13: 0x3fff707c
       A14: 0x00000000       A15: 0x3fff6e90       SAR: 0x00000004  EXCCAUSE: 0x0000001c

```

## Project Include elf and sdkconfig

https://send.firefox.com/download/5aa866a32947a9c1/#8mHzcV1xFABwv7QPYxeKng

donghengqaz, 2019-04-08:
这是有空指针传递进去了吧。

donghengqaz, 2019-04-08:
你的工程怎么操作？我试了 patch 打不上。

FayeY, 2019-04-17:
想请问，这个问题有什么进展吗？

nickfox-taterli, 2019-04-23:
空指针没有传递,只要不涉及snprintf进行%f修饰,就不会有问题,进行%d修饰同样的变量,没有问题.

donghengqaz, 2019-04-28:
我试下使用你的代码没有什么问题，最近解决了一些踩内存的问题，你可以试试 master 分支。
```
int HAL_Snprintf(char *str, const int len, const char *fmt, ...)
{
    va_list args;
    int rc;

    va_start(args, fmt);
    rc = vsnprintf(str, len, fmt, args);
    va_end(args);

    return rc;
}

/******************************************************************************
 * FunctionName : app_main
 * Description  : entry of user application, init user function here
 * Parameters   : none
 * Returns      : none
*******************************************************************************/
void app_main(void)
{
    char buf[128];

    HAL_Snprintf(buf, 128, "%s, %f\n", esp_get_idf_version(), 3.1415926);

    printf("SDK version:%s\n", buf);
}
```

donghengqaz, 2019-04-28:
Log 如下：
```
rom BLK0 of EFUSE
I (535) phy_init: phy ver: 1055_12
I (535) reset_reason: RTC reset 1 wakeup 0 store 0, reason is 1
SDK version:v3.2-dev-300-gb2f76cd9-dirty, 3.141593
```

FayeY, 2019-05-06:
由于长时间无更新，此问题暂时关闭；如果最新 master 仍有问题，欢迎 reopen issue。

closed 2019-05-06 by FayeY