espressif/ESP8266_RTOS_SDK #549: Failed to find Xtensa toolchain (IDFGH-927)

ghmeyes, 2019-04-08:
hello:
    I follow the steps of Standard Setup of Toolchain for Linux
Install Prerequisites,but failed.
   
   my system is  ESP8266_lubuntu_20141021.ova

   esp8266@esp8266-VirtualBox:~/esp/project_template$ printenv PATH
/opt/xtensa-lx106-elf/bin:/home/esp8266/esp/xtensa-lx106-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/esp8266/esp/xtensa-lx106-elf/bin:/home/esp8266/esp/xtensa-lx106-elf/bin
esp8266@esp8266-VirtualBox:~/esp/project_template$ make
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
LD build/bootloader/bootloader.elf
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lstdc++
collect2: error: ld returned 1 exit status
make[1]: *** [/home/esp8266/esp/project_template/build/bootloader/bootloader.elf] Error 1
make: *** [/home/esp8266/esp/project_template/build/bootloader/bootloader.bin] Error 2
esp8266@esp8266-VirtualBox:~/esp/project_template$

ghmeyes, 2019-04-08:
sorry ,I got it!@igrr

donghengqaz, 2019-04-09:
If you use ESP8266_RTOS_SDK v3.1 and later version, please use the newer lx106 toolchian, [for x64 linux](https://dl.espressif.com/dl/xtensa-lx106-elf-linux64-1.22.0-88-gde0bdc1-4.8.5.tar.gz) or [for x86 linux](https://dl.espressif.com/dl/xtensa-lx106-elf-linux32-1.22.0-88-gde0bdc1-4.8.5.tar.gz).

FayeY, 2019-04-17:
Is this issue solved?

closed 2019-04-24 by FayeY

leejaeuk, 2019-05-24:
i suffer same problem
how can i solve this problem?