espressif/ESP8266_RTOS_SDK #550: Can the PWM APIs support to generate 15KHz signal or larger ?

kmalloc8, 2019-04-09:
The PWM maximum is 1KHz, how to generate a larger signal?

xiongyumail, 2019-04-09:
Need duty cycle change?

kmalloc8, 2019-04-09:
@xiongyumail , yes, support adjust the duty also

xiongyumail, 2019-04-09:
You can try sigama_delta, but when the duty cycle is adjustable, it has a fixed frequency of 315 KHz and only one channel.

kmalloc8, 2019-04-09:
@xiongyumail , Do you have examples how to use the sigama_delta?

xiongyumail, 2019-04-09:
```c
void sigma_delta_init()
{
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDI_U,FUNC_GPIO12);
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTMS_U,FUNC_GPIO14);
    GPIO_OUTPUT_SET(12, 1);
    GPIO_OUTPUT_SET(14, 1);
    GPIO_REG_WRITE(GPIO_PIN12_ADDRESS,1); // connect GPIO 12 to Sigmadelta Channel
    GPIO_REG_WRITE(GPIO_PIN14_ADDRESS,1); // connect GPIO 14 to Sigmadelta Channel
}
```

```c

void sigama_delta_set_carrier(float carrier_duty)
{
    uint8_t target = 0,prescale=0;
    uint8_t duty = (uint8_t)(255 - carrier_duty*255);
    target = (duty>128)?(256-duty):duty;
    prescale = (target==0)?0:(target-1);

    GPIO_REG_WRITE(GPIO_SIGMA_DELTA_ADDRESS, 0x10000 | (prescale<<8) | duty); // set Target Prescaler and duty cycle
}
```

kmalloc8, 2019-04-09:
@xiongyumail ,Thanks.

closed 2019-04-17 by FayeY