espressif/ESP8266_RTOS_SDK #555: Please tell me how i can get the smartConfigType from the callBack funtion ？

xuhongv, 2019-04-13:
- now I can use airkiss type to  connects to AP  ，but I don't kown current smartconfig type from the callback funtion  ：  

```
 case SC_STATUS_LINK_OVER:
        ESP_LOGI(TAG, "SC_STATUS_LINK_OVER");
        // rtos 2.0 sdk 
        if (pdata != NULL) {
            uint8_t phone_ip[4] = {0};
            memcpy(phone_ip, (uint8_t *)pdata, 4);
            ESP_LOGI(TAG, "lcoal phone ip : %d.%d.%d.%d ", phone_ip[0], phone_ip[1], phone_ip[2], phone_ip[3]);
        }  else  {
            ESP_LOGI(TAG, "airkiss start");
        }
        xEventGroupSetBits(wifi_event_group, ESPTOUCH_DONE_BIT);
        break;
```

xuhongv, 2019-04-13:
[image]

ustccw, 2019-04-15:
by `esp_smartconfig_set_type()` API to set type

xuhongv, 2019-04-15:
- en。我已经设置为airkiss配网，并且可以配网成功，但是 回调中无法判断是否 airkiss还是 esptouch类型。

xuhongv, 2019-04-15:
- 因为我要通过本次配网是否为 `airkiss`方式，从而做我自己的业务。

ustccw, 2019-04-15:
既然你都设置了 airkiss, 那 callback 里就是 airkiss 的. 这两个是一致的.

xuhongv, 2019-04-15:
- 可能您没用理解我意思！我现设置的是  ` esp_smartconfig_set_type(SC_TYPE_ESPTOUCH_AIRKISS);` ，我希望我的程序支持airkiss 和 esptouch，而我是从回调获取是哪种配网方式！

ustccw, 2019-04-15:
当前 SDK 回调函数里, 是没法知道哪种配网方式的.

xuhongv, 2019-04-15:
嗯嗯。我希望贵方可以把这个完善下。
因为 rtos2.0或者nonos2.0都是支持的！而且我上方的截图都是显示 airkiss类型，说明你们在处理这个spinff嗅探时候，已经是有判断出类型了，只是回调给我们这层的逻辑没做好！希望完善下就可以了！

ustccw, 2019-04-15:
截图显示 airkiss 不是通过识别 sniffer 包的, 而是通过 `esp_smartconfig_set_type` 读取后, 决定采用哪种配网方式.

ustccw, 2019-04-15:
默认的是 airkiss

xuhongv, 2019-04-15:
- 那么既然之前可以回调，现在不行吗？这个问题，贵方有没有准备解决下呢？

ustccw, 2019-04-15:
没明白你详细的需求. `程序支持 airkiss 和 esptouch`, 当前同一时间, 只能有一种配网方式.

xuhongv, 2019-04-15:
- 我的业务是支持2种配网方式，并不是同时去用这2个方式去配网！
- 这样把！我打个比方！用户现在用的是 esptouch 配网，下次用的是 微信airkiss配网 ！

xuhongv, 2019-04-15:
- 我现在遇到的问题是：按照以往贵方提供的配网回调中，可以通过 `pdata` 是否为空从而判断是当前哪种配网模式！但是目前 rtos3.1 ，不管是airkiss或者是esptouch，这个 `pdata`都是空！

[image]

xuhongv, 2019-04-15:
- 现在就是想咨询下，有木有计划把这个完善下？或者是给我点建议

ustccw, 2019-04-15:
如果第一次用的是 airkiss, 下一次用 esptouch, 那程序中会调用 `esp_smartconfig_set_type` 来切换配网方式的. 否则都是默认的 airkiss.

xuhongv, 2019-04-15:
- 我现在的工程逻辑是 每次用户物理按钮配网标志位写一，然后设备重启读取配网标志位后是否为配网流程，再配网标志位写0！所以都是默认的 airkiss 把？
- 我主要想咨询下，有没有办法可以从 配网回调函数中知道当前的配网方式！

ustccw, 2019-04-15:
1. 是的, 默认就是 `airkiss`. 
2. 当前配网回调函数中, 是无法知道当前的配网方式. 以前通过 `pdata` 来判断也是比较业余的做法.

xuhongv, 2019-04-15:
- 那么我配网前设置 esp_smartconfig_set_type(SC_TYPE_ESPTOUCH_AIRKISS); ，底层数据打印中可以准确看到当前是 esptouch 还是 airkiss 方式！为什么不把这个方式也回调给我们应用层呢？

FireWinter, 2019-05-06:
> * 那么我配网前设置 esp_smartconfig_set_type(SC_TYPE_ESPTOUCH_AIRKISS); ，底层数据打印中可以准确看到当前是 esptouch 还是 airkiss 方式！为什么不把这个方式也回调给我们应用层呢？

后续版本会将配网方式传到应用层，请关注版本更新~

FayeY, 2019-06-27:
#544

closed 2019-06-27 by FayeY