espressif/ESP8266_RTOS_SDK #558: 模块服务后停在 read default base MAC address from BLK0 of E

wanghui0574xs, 2019-04-16:
--- idf_monitor on /dev/cu.SLAB_USBtoUART 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16
0x40100000: _stext at ??:?

tail 12
chksum 0xcb
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12
tail 12
chksum 0xe2
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12
tail 0
chksum 0x80
I (67) boot: ESP-IDF v3.2-dev-182-g94f78960 2nd stage bootloader
I (67) boot: compile time 19:16:31
I (76) qio_mode: Enabling default flash chip QIO
I (76) boot: SPI Speed      : 40MHz
I (82) boot: SPI Mode       : QIO
I (88) boot: SPI Flash Size : 8MB
I (94) boot: Partition Table:
I (100) boot: ## Label            Usage          Type ST Offset   Length
I (111) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (123) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (135) boot:  2 factory          factory app      00 00 00010000 000f0000
I (146) boot: End of partition table
I (153) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x2c0f8 (180472) map
I (226) esp_image: segment 1: paddr=0x0003c110 vaddr=0x3ffe8000 size=0x0050c (  1292) load
I (227) esp_image: segment 2: paddr=0x0003c624 vaddr=0x3ffe850c size=0x0011c (   284) load
I (238) esp_image: segment 3: paddr=0x0003c748 vaddr=0x40100000 size=0x057f8 ( 22520) load
0x40100000: _stext at ??:?

I (258) boot: Loaded app from partition at offset 0x10000
I (300) system_api: Base MAC address is not set, read default base MAC address from BLK0 of E

wanghui0574xs, 2019-04-17:
发现base_gpio_init()执行后, 就卡住了.
assert(base_gpio_init() == 0);

wanghui0574xs, 2019-04-17:
在startup.c 里的user_init_entry()函数里, base_gpio_init()会导致上述问题出现

wanghui0574xs, 2019-04-17:
base_gpio_init()  这个函数没有找到定义. 这个函数是否一定要执行. ?

donghengqaz, 2019-04-19:
这个函数会初始化一些特殊的 GPIO，理论上可以不要。

closed 2019-05-06 by FayeY