espressif/ESP8266_RTOS_SDK #561: 当我在程序中设置每次上电后PWM占空比为0，但是依然前1s一直有高电平输出，之后恢复占空比为0？

xuhongv, 2019-04-19:
## 环境

- Development Kit:
- IDF version  SDK version:v3.1-43-gd7da7bd9
- Development Env:   Make
- Operating System:     Windows
- Power Supply:         external 3.3V

### 我的代码如下：

```
#define PWM_0_OUT_IO 0
#define PWM_1_OUT_IO 3

// PWM period 500us(2Khz), same as depth
#define PWM_PERIODS (500)
// pwm pin number
const uint32_t pin_nums[2] = { PWM_0_OUT_IO_NUM, PWM_1_OUT_IO_NUM};
// dutys table, (duty/PERIOD)*depth
uint32_t mDuties[2] = {0,0};
// phase table, (phase/180)*depth
int16_t phases[4] = {   0,  0};

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    //获取IDF版本
    printf("     SDK version:%s\n", esp_get_idf_version());
    pwm_init(PWM_PERIODS, mDuties, 2, pin_nums);
    pwm_set_phases(phases);
    pwm_start();
}
```

## 疑问

- [ ] 我已经在程序入口`app_main()`中设置pwm的占空比为0 了，**为什么每次上电前1s左右先是高电平然后再到我自己的设置？**
- [ ] 我之前使用的 rtos2.0 版本不会出现这个问题的？难道我要修改硬件电路？

redchenjs, 2019-04-19:
硬件没有下拉吗？

xuhongv, 2019-04-19:
- 没下拉的。

xiongyumail, 2019-04-19:
@xuhongv 
1. 是指PWM初始化后一秒还有高电平输出嘛？
2. 我看到你使用了IO3, 是不需要串口接收功能嘛？

xuhongv, 2019-04-19:
- 对！初始化前一秒内还是有高电平的输出！我使用 io3 这个特殊的接收管脚是因为项目需要！目前程序没用接收功能的！

xuhongv, 2019-04-19:
- 我硬件不能设置下拉，如果我的pwm脚 io0 下拉，就进去烧录模式了。

xiongyumail, 2019-04-19:
你可以先尝试先置低对应io，再初始化PWM

xuhongv, 2019-04-19:
- 感谢！我只能这样先尝试了！后面待你们修复下这个问题，我再修改把！

xuhongv, 2019-04-19:
- 目前我代码如下，依然是上电后先**高电平再低电平**，我只能回退到rtos2.0了？？

```
#define PWM_0_OUT_IO 0
#define PWM_1_OUT_IO 3

// PWM period 500us(2Khz), same as depth
#define PWM_PERIODS (500)
// pwm pin number
const uint32_t pin_nums[2] = {PWM_0_OUT_IO_NUM, PWM_1_OUT_IO_NUM};
// dutys table, (duty/PERIOD)*depth
uint32_t mDuties[2] = {0, 0};
// phase table, (phase/180)*depth
int16_t phases[4] = {0, 0};

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    //获取IDF版本
    printf("SDK version:%s\n", esp_get_idf_version());

    //输出模式，禁止中断
    gpio_config_t io_conf;
    //禁止中断
    io_conf.intr_type = GPIO_INTR_DISABLE;
    //设置为输出模式
    io_conf.mode = GPIO_MODE_OUTPUT;
    //管脚的位
    io_conf.pin_bit_mask = ((1ULL << PWM_0_OUT_IO) | (1ULL << PWM_1_OUT_IO));
    //禁止下拉
    io_conf.pull_down_en = 0;
    //禁止上拉
    io_conf.pull_up_en = 0;
    //开始配置管脚
    gpio_config(&io_conf);

    gpio_set_level(PWM_0_OUT_IO, 0);
    gpio_set_level(PWM_1_OUT_IO, 0);

    pwm_init(PWM_PERIODS, mDuties, 2, pin_nums);
    pwm_set_phases(phases);
    pwm_start();
}
```

xiongyumail, 2019-04-19:
能否看下示波器或逻辑分析仪波形？

xiongyumail, 2019-04-19:
@xuhongv 
我这里测试，PWM初始化到PWM信号出来只有50ms左右
[image]

[image]

xuhongv, 2019-04-19:
- 我这里使用的接入LED灯具，每次上电都会**先是全亮**再到我想要的效果，这很不友好！我建议你接个led看最明显了！

xiongyumail, 2019-04-19:
@xuhongv 
ESP8266 上电初始化需要一定时间，GPIO0不建议做PWM输出，GPIO0上电会有一段时间输出CLK

xuhongv, 2019-04-19:
- 那我先回退用 Rtos2.0 了。

closed 2019-04-19 by xuhongv

leovs, 2019-05-13:
想做个渐变灯，为什么为用示例代码，pwm并没有效果？  io口用的13
pwm_start   gpio口电压直接变 3.3  stop后 电压降到0  中间没有任何渐变过程？

xiongyumail, 2019-05-13:
@leovs 
渐变需要应用调节占空比， PWM驱动只是根据参数产生对应脉宽，频率的波形。

leovs, 2019-05-25:
> @leovs
> 渐变需要应用调节占空比， PWM驱动只是根据参数产生对应脉宽，频率的波形。

用你上面的代码
万用表测电压，gnd 和 13  电压是0
vcc 和13  电压是3.3 ，一直不变

用官方的代码
会从3.3   20秒后直接变为0  20秒后直接变成3.3

我理解脉宽是电压持续事件，波形应该是电压变化，pwm应该是波形，非方波， 目前我的现象看起来是方波。

leovs, 2019-05-25:
#define PWM_0_OUT_IO_NUM   12
#define PWM_1_OUT_IO_NUM   13
#define PWM_2_OUT_IO_NUM   14
#define PWM_3_OUT_IO_NUM   15

// PWM period 500us(2Khz), same as depth
#define PWM_PERIOD    (500)

static const char *TAG = "pwm_example";

// pwm pin number
const uint32_t pin_num[4] = {
    PWM_0_OUT_IO_NUM,
    PWM_1_OUT_IO_NUM,
    PWM_2_OUT_IO_NUM,
    PWM_3_OUT_IO_NUM
};

// duties table, real_duty = duties[x]/PERIOD
uint32_t duties[4] = {
    250, 250, 250, 250,
};

// phase table, delay = (phase[x]/360)*PERIOD
int16_t phase[4] = {
    0, 0, 90, -90,
};

void app_main()
{
    pwm_init(PWM_PERIOD, duties, 4, pin_num);
    pwm_set_channel_invert(0x1 << 0);
    pwm_set_phases(phase);
    pwm_start();
    int16_t count = 0;

    while (1) {
        if (count == 20) {
            // channel0, 1 output hight level.
            // channel2, 3 output low level.
            pwm_stop(0x3);
            ESP_LOGI(TAG, "PWM stop\n");
        } else if (count == 30) {
            pwm_start();
            ESP_LOGI(TAG, "PWM re-start\n");
            count = 0;
        }

        count++;
        vTaskDelay(1000 / portTICK_RATE_MS);
    }
}

官方代码， 不知道怎样能生成正旋波