espressif/ESP8266_RTOS_SDK #562: 如何设置主机名称 tcpip_adapter_set_hostname一直返回错误.

wanghui0574xs, 2019-04-22:


redchenjs, 2019-04-22:
同问

donghengqaz, 2019-04-28:
需要在某个模式（sta/ap）启动以后才能配置 hostname，代码如下：
```
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    int ret;

    switch (event->event_id) {
        case SYSTEM_EVENT_STA_START:
            ret = tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, "espressif");
            printf("tcpip_adapter_set_hostname result %x\n", ret);
            esp_wifi_connect();
            break;

        case SYSTEM_EVENT_STA_GOT_IP:
            xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
            break;

        case SYSTEM_EVENT_STA_DISCONNECTED:
            /* This is a workaround as ESP32 WiFi libs don't currently
               auto-reassociate. */
            esp_wifi_connect();
            xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
            break;

        default:
            break;
    }

    return ESP_OK;
}
```

redchenjs, 2019-04-28:
@donghengqaz 谢谢，将设置hostname函数放到WiFi回调里面后测试可行。

closed 2019-05-06 by FayeY