espressif/ESP8266_RTOS_SDK #565: Recommended to use the latest esp-mqtt 推荐使用最新的 esp-mqtt

FayeY, 2019-04-24:
It is suggested to use our [esp-mqtt](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/protocols/esp-mqtt) which also support SSL MQTT. The esp-mqtt will be the one we mainly focus on in future development.

推荐使用我们最新的 [esp-mqtt](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/protocols/esp-mqtt)，它支持了 SSL MQTT，会是后续我们主要维护的 MQTT 版本。

请知悉。谢谢！

microponts, 2019-04-24:
新的工程跑起来卡在了I (238) boot: Loaded app from partition at offset 0x10000 这里，是什么原因呀？

ustccw, 2019-04-24:
@microponts 先擦除一下 flash 后, 再试试看呢.
```
make erase_flash && make flash monitor
```

wanghui0574xs, 2019-04-30:
esp-aliyun 这个项目是用你们的esp-mqtt么?

microponts, 2019-04-30:
@FayeY 擦除了flash后 烧录进去还是卡在boot这里

                                                                                         lI (43) boot: ESP-IDF  2nd stage bootloader
I (43) boot: compile time 23:15:54
I (43) qio_mode: Enabling default flash chip QIO
I (43) boot: SPI Speed      : 80MHz
I (46) boot: SPI Mode       : QIO
I (49) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (68) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  3 factory          factory app      00 00 00010000 00200000
I (87) boot:  4 storage          Unknown data     01 82 00210000 001ec000
I (94) boot: End of partition table
I (97) boot: Defaulting to factory image
I (101) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x5cc4c (379980) map
I (217) esp_image: segment 1: paddr=0x0006cc64 vaddr=0x3ffe8000 size=0x0068c (  1676) load
I (218) esp_image: segment 2: paddr=0x0006d2f8 vaddr=0x3ffe868c size=0x00174 (   372) load
I (222) esp_image: segment 3: paddr=0x0006d474 vaddr=0x40100000 size=0x06820 ( 26656) load
0x40100000: _stext at ??:?

I (238) boot: Loaded app from partition at offset 0x10000

----------------下面是我的配置----------------
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#
CONFIG_IDF_TARGET_ESP32=
CONFIG_IDF_TARGET_ESP8266=y

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-lx106-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_BOOTLOADER_INIT_SPI_FLASH=y
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_CHECK_APP_SUM=y
CONFIG_BOOTLOADER_CHECK_APP_HASH=
CONFIG_BOOTLOADER_APP_TEST=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/tty.SLAB_USBtoUART"
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=y
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=
[... 423 lines cut ...]
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_ADAPTER_GLOBAL_DATA_LINK_IRAM=y

#
# Util
#
CONFIG_ESP_SHA=y
CONFIG_ESP_AES=y

#
# Virtual file system
#
CONFIG_USING_ESP_VFS=y
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Unified Provisioning
#
CONFIG_ENABLE_UNIFIED_PROVISIONING=

#
# wpa_supplicant
#
CONFIG_LTM_FAST=y

microponts, 2019-04-30:
@FayeY hi,我试了一下，发现主干的代码是能正常编译跑起来的。但是我只要勾选了spiffs选项后，跑起来就卡在了I (249) boot: Loaded app from partition at offset 0x10000 boot加载程序这里。之后就算去掉了spiffs选项后，编译跑起来 依旧卡 I (249) boot: Loaded app from partition at offset 0x10000 boot ,会不会跟分区表有关？
-------下面是我的分区表-------
nvs,      data, nvs,     0x9000,	0x4000,
otadata,  data, ota,     0xd000,	0x2000
phy_init, data, phy,     0xf000,	0x1000,
factory,  app,  factory, 0x10000,	3M,
storage,  data, spiffs,  0x310000,	0xEC000,

xuhongv, 2019-05-06:
@FayeY 请问这个 mqtt 组件啥时候可以更新到 3.1分支

FayeY, 2019-05-20:
@xuhongv ，你好，
原则上来说，对旧版本只做 bug fix 更新，不考虑增加新功能。建议可以后续使用我们发布的新版本。
如果给您带来不便，深感抱歉。
请知悉。谢谢！

sonwa, 2019-07-20:
@FayeY 请问用着用着时不时就会挂掉，一般是什么引起的？SDK用的是3.2

Task stack [tiT] stack from [0x3fff1e00] to [0x3fff25fc], total [2048] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c 

3fff2540  0x00000060 0x40108508 0x401083d4 0x4023334d 0x40108508 0x40108508 0x40108508 0x40106e28 0x40105e3c 0x4010841c 0x4010841c 0x4024d49a 0x40105e3c 0x4010841c 0x40108508 0x40237674 
0x4023334d: sys_mbox_trypost at /Users/sw/esp/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_arch.c:392

0x40105e3c: xthal_set_intset at ??:?

0x4024d49a: recv_tcp at /Users/sw/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/api/api_msg.c:1562 (discriminator 6)

0x40105e3c: xthal_set_intset at ??:?

0x40237674: tcp_input at /Users/sw/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/tcp_in.c:435 (discriminator 1)

3fff2580  0x3ffec60c 0x40108508 0x40106e28 0x40232a0d 0x00000001 0x00000014 0x3fff25e0 0x40233399 0x402115f8 0x40230264 0x4010593c 0x40106e28 0x00000096 0x00000008 0x40108508 0x40233169 
0x40232a0d: ip4_input at /Users/sw/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/ip4.c:685

0x40233399: sys_arch_mbox_fetch at /Users/sw/esp/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_arch.c:392

0x40230264: cyclic_timer at /Users/sw/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/timeouts.c:387

0x4010593c: xthal_set_intset at ??:?

0x40233169: ethernet_input at /Users/sw/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/netif/ethernet.c:182

3fff25c0  0x00000000 0x3fff25e0 0x4010593c 0x00000000 0x00000000 0x00000000 0x3fff3420 0x402339ad 0x4010852c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 
0x4010593c: xthal_set_intset at ??:?

0x402339ad: tcpip_thread at /Users/sw/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/api/tcpip.c:459

        PC: 0x4010bffb        PS: 0x00000030        A0: 0x4024d4ac        A1: 0x3fff2570
0x4024d4ac: recv_tcp at /Users/sw/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/api/api_msg.c:1562 (discriminator 6)

        A2: 0x401083d4        A3: 0x00000000        A4: 0x00000060        A5: 0x4010bff8
        A6: 0x40108508        A7: 0x00000000        A8: 0x00000020        A9: 0x00000001
       A10: 0x00000014       A11: 0x3ffec634       A12: 0x401083d4       A13: 0x00000000
       A14: 0x00000060       A15: 0x40106e28       SAR: 0x0000001d  EXCCAUSE: 0x00000000

Task stack [tiT] stack from [0x3fff1e00] to [0x3fff25fc], total [2048] size

register map is 3ffed940 error
                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c 

3fff1e00  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff1e40  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff1e80  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff1ec0  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff1f00  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff1f40  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff1f80  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff1fc0  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff2000  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff2040  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff2080  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff20c0  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff2100  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff2140  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
[... cut ...]
        A6: 0x00000000        A7: 0x40211428        A8: 0x40211510        A9: 0x00000001
       A10: 0x00000014       A11: 0x3ffec634       A12: 0x3ffe92ec       A13: 0x00000000
       A14: 0x0000001c       A15: 0x00000008       SAR: 0x0000001e  EXCCAUSE: 0x00000002
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7204, room 16 
0x40100000: _stext at ??:?

tail 4
chksum 0xa1
load 0x3ffe8408, len 24, room 4 
tail 4
chksum 0x28
load 0x3ffe8420, len 3548, room 4 
tail 8
chksum 0x16

kmalloc8, 2019-07-23:
新版MQTT非常容易掉线,何解"**E (121975) MQTT_CLIENT: No PING_RESP, disconnected**", 
而2.0SDK版本同样网络条件下没问题啊

sonwa, 2019-07-23:
对，会经常掉线。严重的时候会自动重启，收发几十个数据后一般都会掉一次线

Jerry-zo, 2019-10-25:
我的mqtt也经常掉线" E (2889055) MQTT_CLIENT: No PING_RESP, disconnected",我以为是心跳问题,但mqtt_config.keepalive如何设置都会掉.

wanghui0574xs, 2019-10-25:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    我使用阿里云的SDK, 这个SDK在linux主机里, 不会掉线. 我把阿里云的sdk一直到ESP8266_RTOS_SDK上, 还是会掉线. 估计跟ESP8266_RTOS_SDK 有关. 可能是网络不稳定导致的.
    
        
            
        
    
    
          
        
            
    
        font{
            line-height: 1.5;
        }
    
----------------------------wanghui0574xs@sina.com

sky9971, 2019-11-01:
> 我的mqtt也经常掉线" E (2889055) MQTT_CLIENT: No PING_RESP, disconnected",我以为是心跳问题,但mqtt_config.keepalive如何设置都会掉.

@Jerry-zo ,
I'm also facing this issue. Have you been able to find a fix for this issue?
In my case it drops after publishing a few messages and fails to reconnect. 
Mbedtls throws -0x4310 error

kmalloc8, 2020-01-14:
It always happen..., how to fix this problem 

`MQTT_CLIENT: Error write data or timeout, written len = 0`

wanghui0574xs, 2020-01-14:
你写了一个空帧吧 ---------------------------- wanghui0574xs@sina.com 在2020年01月14日 15:47，Gavin 写道： It always happen..., how to fix this problem MQTT_CLIENT: Error write data or timeout, written len = 0

kmalloc8, 2020-01-14:
@wanghui0574xs , 没有空桢，不是每次都是出现这个，网速可能也不好，但是出现这个也太多了，mqtt老断，publish不成功。因为我要定时上报数据，非常容易出现mqtt disconnected。这个esp-mqtt的正确使用流程是啥？

wanghui0574xs, 2020-01-14:
len=0，你的内容为空了吧，mqtt部分，我用了阿里云的SDK ---------------------------- wanghui0574xs@sina.com 在2020年01月14日 22:05，Gavin 写道： @wanghui0574xs , 没有空桢，不是每次都是出现这个，网速可能也不好，但是出现这个也太多了，mqtt老断，publish不成功。因为我要定时上报数据，非常容易出现mqtt disconnected。这个esp-mqtt的正确使用流程是啥？

nechet, 2020-03-17:
Hello, 
Please help.
MQTT example does not work for me:
- branch: master or release/v3.3
- ESP8266_RTOS_SDK/examples/protocols/mqtt/tcp/
(MQTT example from branch: release/v3.3 and ESP8266_RTOS_SDK/examples/protocols/ibm-mqtt/ works well for me in same time)
`
user@pc:~/bin/espressif/ESP8266_RTOS_SDK/examples/protocols/mqtt/tcp$ make -j4 flash
Toolchain path: /home/user/bin/espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-100-ge567ec7
Compiler version: 5.2.0
Python requirements from /home/user/bin/espressif/ESP8266_RTOS_SDK/requirements.txt are satisfied.
Flashing binaries to serial port /dev/ttyUSB1 (app at offset 0x10000)...
esptool.py v2.4.0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:a0:f3:13
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 9616 bytes to 6785...
Wrote 9616 bytes (6785 compressed) at 0x00000000 in 0.6 seconds (effective 128.2 kbit/s)...
Hash of data verified.
Compressed 499216 bytes to 340424...
Wrote 499216 bytes (340424 compressed) at 0x00010000 in 30.0 seconds (effective 133.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 83...
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.0 seconds (effective 2042.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
nechet@andrey-pc:~/bin/espressif/ESP8266_RTOS_SDK/examples/protocols/mqtt/tcp$ make monitor
Toolchain path: /home/user/bin/espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-100-ge567ec7
Compiler version: 5.2.0
Python requirements from /home/user/bin/espressif/ESP8266_RTOS_SDK/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB1 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

`