espressif/ESP8266_RTOS_SDK #567: tcpip_adapter_recv_cb memory leak?

ourairquality, 2019-04-25:
tcpip_adapter_recv_cb()  includes:
```
p = os_malloc(sizeof(struct tcpip_adapter_pbuf));
```
but if pbuf_alloced_custom() fails and returns NULL then the function returns without freeing the above allocation, and might that be a memory leak?

While looking at this code, does the caller free the passed rx buffer if tcpip_adapter_recv_cb() returns ENOMEM or are those paths leaking an rx buffer?

ourairquality, 2019-04-25:
While checking this, could you please also check if the called function `ethernetif_input()` should be freeing the pbuf along the exit path for the case `netif == NULL`, as it does for `p->payload == NULL` etc? It appears that it should free the pbuf in this case too, as `tcpip_adapter_recv_cb()` does not free the pbuf in that case, not in any cases.

donghengqaz, 2019-04-28:
Thanks your support, I will fix these bugs as soon as possible. 

If `tcpip_adapter_recv_cb` return not 0, internal function will free the "buffer".

FayeY, 2019-05-20:
Please have a try with the latest master, it should have solved this issue. Thanks.

closed 2019-05-20 by FayeY