espressif/ESP8266_RTOS_SDK #570: 无法连接wifi, SYSTEM_EVENT_STA_DISCONNECTED 断网提示里的reason值哪里有解释

wanghui0574xs, 2019-04-29:
原先能正常联网的程序, 最近不知为何就无法联网了. 
最先发生在一个程序能正常联网, 建立mqtt连接. 接着网关关掉. 模块断网, 持续一晚上后, 开启网关, 模块就无法联网.
现在程序一直无法联网. 
**### _SYSTEM_EVENT_STA_DISCONNECTED 断网提示里的reason值, 在那份文档里有说明?_**
state: 5 -> 0 (200)
rm 0
pm close 7 0 0/10001588
D (36157) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:TP-LINK_Edgesense, ssid_len:17, bssid:88:25:93:5c:d8:af, reason:204
V (36167) event: enter default callback
D (36167) tcpip_adapter: if0 start ip lost tmr: enter
D (36177) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x40107b58 interval=120 ip=0
V (36197) event: exit default callback

scandone
state: 0 -> 2 (b0)

state: 2 -> 0 (200)
D (37317) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:TP-LINK_Edgesense, ssid_len:17, bssid:88:25:93:5c:d8:af, reason:2
V (37327) event: enter default callback
D (37327) tcpip_adapter: if0 start ip lost tmr: enter
D (37337) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x40107b58 interval=120 ip=0
V (37347) event: exit default callback

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)

state: 3 -> 0 (400)
D (38487) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:TP-LINK_Edgesense, ssid_len:17, bssid:88:25:93:5c:d8:af, reason:4
V (38487) event: enter default callback
D (38497) tcpip_adapter: if0 start ip lost tmr: enter
D (38497) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x40107b58 interval=120 ip=0
V (38517) event: exit default callback

wanghui0574xs, 2019-04-30:
scandone

D (33095) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:1, ssid_len:1, bssid:00:00:00:00:00:00, reason:201
V (33095) event: enter default callback
D (33105) tcpip_adapter: if0 start ip lost tmr: enter
D (33105) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x40107d50 interval=120 ip=0
V (33125) event: exit default callback

wanghui0574xs, 2019-04-30:
scandone
state: 0 -> 2 (b0)

state: 2 -> 0 (200)

D (69538) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:HUAWEI Mate 8, ssid_len:13, bssid:60:83:34:0b:36:bc, reason:2
V (69538) event: enter default callback
D (69548) tcpip_adapter: if0 start ip lost tmr: enter
D (69548) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x4010875c interval=120 ip=0
V (69568) event: exit default callback

wanghui0574xs, 2019-04-30:
有时候艰难的连接上了, 但是DHCP却一直分配不出地址
scandone
state: 0 -> 2 (b0)

state: 2 -> 3 (0)

state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt

c
onnected with HUAWEI Mate 8, channel 11
D (88418) event: SYSTEM_EVENT_STA_CONNECTED, ssid:HUAWEI Mate 8, ssid_len:13, bssid:60:83:34:0b:36:bc, channel:11, authmode:1073667484
V (88418) event: enter default callback
D (88428) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (88438) tcpip_adapter: if0 start ip lost tmr: enter
D (88448) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x4010875c interval=120 ip=0
D (88458) udp_sync: UDP sync cache udp 0x4010891c
D (88468) udp_sync: UDP sync clear udp 0x4010891c
D (88468) tcpip_adapter: dhcp client start successfully
V (88478) event: exit default callback

D (88968) tcpip_adapter: dhcpc time(ms): 0

D (89468) tcpip_adapter: dhcpc time(ms): 500

D (89968) tcpip_adapter: dhcpc time(ms): 1000

D (90418) udp_sync: UDP sync cache udp 0x4010891c
D (90418) udp_sync: UDP sync clear udp 0x4010891c

D (90468) tcpip_adapter: dhcpc time(ms): 1500

FayeY, 2019-07-15:
能否用最新发布的 v3.2 试试看是否还有问题？

closed 2019-08-22 by FayeY