espressif/ESP8266_RTOS_SDK #572: Fail to open HTTP connection in OTA example

boriss2oo6, 2019-05-01:
## Environment

- Development Kit: Wemos D1 mini
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    Latest (2b579ef758a9fc88f76b70cf849652a352dcc555)
- Development Env:      Visual Code / PlatformIo
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

E (5885) esp_https_ota: Failed to open HTTP connection: 28675

### Expected Behavior

### Actual Behavior

### Steps to repropduce

Just used provided ota example.

### Code to reproduce this issue

/* OTA example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_ota_ops.h"
#include "esp_http_client.h"
#include "esp_https_ota.h"

#include "nvs.h"
#include "nvs_flash.h"

static const char *TAG = "simple_ota_example";
extern const char server_cert_pem_start[] asm("_binary_src_html_server_pem_start");
//extern const char server_cert_pem_end[] asm("_binary_src_html_server_pem_end");

/* FreeRTOS event group to signal when we are connected & ready to make a request */
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
const int CONNECTED_BIT = BIT0;

esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
    switch(evt->event_id) {
        case HTTP_EVENT_ERROR:
            ESP_LOGD(TAG, "HTTP_EVENT_ERROR");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_CONNECTED");
            break;
        case HTTP_EVENT_HEADER_SENT:
            ESP_LOGD(TAG, "HTTP_EVENT_HEADER_SENT");
            break;
        case HTTP_EVENT_ON_HEADER:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
            break;
        case HTTP_EVENT_ON_DATA:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
            break;
        case HTTP_EVENT_ON_FINISH:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_FINISH");
            break;
        case HTTP_EVENT_DISCONNECTED:
            ESP_LOGD(TAG, "HTTP_EVENT_DISCONNECTED");
            break;
    }
    return ESP_OK;
}

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch (event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        /* This is a workaround as ESP32 WiFi libs don't currently
           auto-reassociate. */
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

static void initialise_wifi(void)
{
    tcpip_adapter_init();
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "DIGI-Peh2",
            .password = "VRjpUfR8",
        },
    };
    ESP_LOGI(TAG, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_start() );
}

void simple_ota_example_task(void * pvParameter)
{
    ESP_LOGI(TAG, "Starting OTA example...");
    
    /* Wait for the callback to set the CONNECTED_BIT in the
       event group.
    */
    xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT,
                        false, true, portMAX_DELAY);
    ESP_LOGI(TAG, "Connect to Wifi ! Start to Connect to Server....");
    
    esp_http_client_config_t config = {
        .url = "https://hostmydream.com/boris/Esp8266/firmware.bin",
        .cert_pem = (char *)server_cert_pem_start,
        .event_handler = _http_event_handler,
    };
    esp_err_t ret = esp_https_ota(&config);
    if (ret == ESP_OK) {
        esp_restart();
    } else {
        ESP_LOGE(TAG, "Firmware Upgrades Failed");
    }
    while (1) {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void app_main()
{
    // Initialize NVS.
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
        // OTA app partition table has a smaller NVS partition size than the non-OTA
        // partition table. This size mismatch may cause NVS initialization to fail.
        // If this happens, we erase NVS partition and initialize NVS again.
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );

    initialise_wifi();
    xTaskCreate(&simple_ota_example_task, "ota_example_task", 8192, NULL, 5, NULL);
}

## Debug Logs

I (1015) simple_ota_example: Setting WiFi configuration SSID DIGI-Peh2...
D (1025) event: SYSTEM_EVENT_STA_START
V (1035) event: enter default callback
V (1035) event: exit default callback
I (1045) simple_ota_example: Starting OTA example...
D (1575) event: SYSTEM_EVENT_STA_CONNECTED, ssid:DIGI-Peh2, ssid_len:9, bssid:60:de:f3:26:38:08, channel:4, authmode:3
V (1575) event: enter default callback
D (1585) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (1585) tcpip_adapter: if0 start ip lost tmr: enter
D (1595) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x40106d24 interval=120 ip=0
D (1615) udp_sync: UDP sync cache udp 0x40106de4
D (1615) udp_sync: UDP sync clear udp 0x40106de4
D (1625) tcpip_adapter: dhcp client start successfully
V (1635) event: exit default callback
D (2125) tcpip_adapter: dhcpc time(ms): 0

D (2625) tcpip_adapter: dhcpc time(ms): 500

D (3125) tcpip_adapter: dhcpc time(ms): 1000

D (3345) udp_sync: UDP sync cache udp 0x40106de4
D (3345) udp_sync: UDP sync clear udp 0x40106de4
D (3625) tcpip_adapter: dhcpc time(ms): 1500

D (3625) udp_sync: UDP sync cache udp 0x40106de4
D (3635) udp_sync: UDP sync clear udp 0x40106de4
D (4125) tcpip_adapter: dhcpc time(ms): 2000

D (4625) tcpip_adapter: if0 dhcpc cb
D (4625) tcpip_adapter: if0 ip changed=1
D (4625) event: SYSTEM_EVENT_STA_GOT_IP, ip:192.168.100.20, mask:255.255.255.0, gw:192.168.100.1
V (4625) event: enter default callback
I (4635) event: sta ip: 192.168.100.20, mask: 255.255.255.0, gw: 192.168.100.1
V (4645) event: exit default callback
I (4655) simple_ota_example: Connect to Wifi ! Start to Connect to Server....
D (4665) HTTP_CLIENT: Begin connect to: https://hostmydream.com:443
D (4665) esp-tls: host:hostmydream.com: strlen 15
D (4675) udp_sync: UDP sync cache udp 0x40107648
D (4685) udp_sync: UDP sync clear udp 0x40107648
D (4695) tcpip_adapter: ip:192.168.100.20,mask:255.255.255.0,gw:192.168.100.1

D (4705) udp_sync: UDP sync close port 43888
D (4805) esp-tls: handshake in progress...
D (5805) simple_ota_example: HTTP_EVENT_ON_CONNECTED
D (5815) HTTP_CLIENT: Write header[3]: GET /boris/Esp8266/firmware.bin HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: hostmydream.com
Content-Length: 0

W (5825) TRANS_SSL: Poll timeout or error, errno=Success, fd=54, timeout_ms=5000
E (5835) HTTP_CLIENT: Error write request
D (5835) simple_ota_example: HTTP_EVENT_DISCONNECTED
D (5885) simple_ota_example: HTTP_EVENT_DISCONNECTED
E (5885) esp_https_ota: Failed to open HTTP connection: 28675
E (5885) simple_ota_example: Firmware Upgrades Failed

donghengqaz, 2019-05-06:
Maybe you can open the LwIP TCP and socket level debug information for this problem, and we can find the solution easily.

boriss2oo6, 2019-05-06:
Ok, so i added to my sdkconfig.h the following:
#define CONFIG_LWIP_DEBUG 1
#define CONFIG_LWIP_SOCKETS_DEBUG 1
#define LWIP_TCP_DEBUG 1
#define LWIP_TCP_INPUT_DEBUG 1
#define LWIP_TCP_FR_DEBUG 1
#define LWIP_TCP_RTO_DEBUG 1
#define LWIP_TCP_CWND_DEBUG 1
#define LWIP_TCP_WND_DEBUG 1
#define LWIP_TCP_OUTPUT_DEBUG 1
#define LWIP_TCP_RST_DEBUG 1
#define LWIP_TCP_QLEN_DEBUG 1
#define LWIP_TCPIP_DEBUG 1
Also i posted the full sdkconfig.h here: https://gist.github.com/boriss2oo6/38d7aad6e2fcf69b31bbcd6bc7ea9caa

And posted the full log here:
https://gist.github.com/boriss2oo6/5492972f00fa33fe8a9bf582047fb1f3

P.S. i need to mention that i saw the same behaviour with the http_client_example, so i belive it's something that i do wrong.

donghengqaz, 2019-05-06:
The bug is that socket multi-thread function has no check socket offset if VFS is enabled, so the "larger" socket ID is treated as an invalid parameter.

donghengqaz, 2019-05-06:
I have pushed the bugfix, and thanks for your report.

boriss2oo6, 2019-05-06:
@donghengqaz Thanks for fast support ! Now i manage to see the new firmware length but the ota still fails, should i open a new issue or we continue here ?
I posted the log here:
https://gist.github.com/boriss2oo6/a9513fbb980cd21e1f6c75a2ec1ea30d

donghengqaz, 2019-05-07:
I see the read result is "read error :-29184:" which is -0x7200, you can configure the SSL RX context buffer length to be larger one at menuconfig:
```
Component config  --->
    SSL  --->
        mbedTLS  --->
            (8192) TLS maximum INPUT message content length
```
Try 8192 or other value.

boriss2oo6, 2019-05-07:
I tried with 8192 and 16384 and still the same issue.

donghengqaz, 2019-05-13:
Do you configure to enable the mbedtls debug at menuconfig ? And I tried the master branch, but no error.

boriss2oo6, 2019-05-13:
@donghengqaz i didn't used menuconfig, i used visual code with platformio extension and the menuconfig is not available. After a clean to project and rebuild i managed to compile.

After your last post i tried lot of combinations and it seems that i can succed OTA only after a full chip erase, seems that if i use the NVS for storing data OTA doesn't work anymore. Another issue is that after the second OTA i encounter a weird situation:

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x3ffe8408, len 24, room 16
tail 8
chksum 0x06
load 0x3ffe8420, len 4256, room 0
tail 0
chksum 0xa1
load 0x40100000, len 7600, room 8
tail 8
chksum 0x93
I (47) boot: ESP-IDF eca84e2 2nd stage bootloader
I (47) boot: compile time 10:02:10
I (47) qio_mode: Enabling default flash chip QIO
D (48) boot: bootloader initialize SPI flash clock and I/O
D (56) boot: magic e9
D (59) boot: segments 03
D (63) boot: spi_mode 00
D (66) boot: spi_speed 00
D (70) boot: spi_size 04
I (73) boot: SPI Speed      : 40MHz
I (78) boot: SPI Mode       : QIO
I (83) boot: SPI Flash Size : 4MB
D (88) boot: mapped partition table 0x8000 at 0x40208000
D (95) flash_parts: partition table verified, 7 entries
I (103) boot: Partition Table:
I (107) boot: ## Label            Usage          Type ST Offset   Length
D (117) boot: load partition table entry 0x3ffffe14
D (124) boot: type=1 subtype=2
I (128) boot:  0 nvs              WiFi data        01 02 00009000 00004000
D (139) boot: load partition table entry 0x3ffffe14
D (146) boot: type=1 subtype=0
I (150) boot:  1 otadata          OTA data         01 00 0000d000 00002000
D (160) boot: load partition table entry 0x3ffffe14
D (167) boot: type=1 subtype=1
I (171) boot:  2 phy_init         RF data          01 01 0000f000 00001000
D (182) boot: load partition table entry 0x3ffffe14
D (189) boot: type=0 subtype=0
I (193) boot:  3 factory          factory app      00 00 00010000 00100000
D (203) boot: load partition table entry 0x3ffffe14
D (210) boot: type=0 subtype=10
I (214) boot:  4 ota_0            OTA app          00 10 00110000 00100000
D (225) boot: load partition table entry 0x3ffffe14
D (232) boot: type=0 subtype=11
I (236) boot:  5 ota_1            OTA app          00 11 00210000 00100000
D (246) boot: load partition table entry 0x3ffffe14
D (253) boot: type=ff subtype=ff
I (258) boot:  6 ÿÿÿÿ¿%,¦ØdÙ0%´Ä<þ?þ?I unknown          ff ff ffffffff ffffffff
I (270) boot: End of partition table
D (275) boot: OTA data offset 0xd000
D (280) boot: OTA sequence values A 0x00000001 B 0x00000002
D (288) boot: Both OTA values valid. Mapping seq 1 -> OTA slot 1
D (297) boot: Trying partition index 1 offs 0x210000 size 0x100000
D (306) esp_image: reading image header @ 0x210000
D (313) esp_image: image header: 0xe9 0x04 0x02 0x02 4022408c
V (321) esp_image: loading segment header 0 at offset 0x210008
V (330) esp_image: segment data length 0x6bcbc data starts 0x210010
I (339) esp_image: segment 0: paddr=0x00210010 vaddr=0x40210010 size=0x6bcbc (441532) map
V (500) esp_image: loading segment header 1 at offset 0x27bccc
V (500) esp_image: segment data length 0x550 data starts 0x27bcd4
I (501) esp_image: segment 1: paddr=0x0027bcd4 vaddr=0x3ffe8000 size=0x00550 (  1360) load
V (513) esp_image: loading segment header 2 at offset 0x27c224
V (521) esp_image: segment data length 0x174 data starts 0x27c22c
I (530) esp_image: segment 2: paddr=0x0027c22c vaddr=0x3ffe8550 size=0x00174 (   372) load
V (543) esp_image: loading segment header 3 at offset 0x27c3a0
V (551) esp_image: segment data length 0x64d0 data starts 0x27c3a8
I (560) esp_image: segment 3: paddr=0x0027c3a8 vaddr=0x40100000 size=0x064d0 ( 25808) load
V (581) esp_image: image start 0x00210000 end of last section 0x00282878
I (582) boot: Loaded app from partition at offset 0x210000
E (590) boot: ERROR: app bin error, start_addr 210000 image_len 469120

donghengqaz, 2019-05-17:
I am sorry that  the application binary must located at place before offset "2MB", and I see you place it at "0x210000". Now the SDK does not support factory binary, so you can remove it and move the 2 OTAs partitions the previous place.

FayeY, 2019-06-27:
Is there any update about this issue？Did the above suggestion help resolve the issue?

boriss2oo6, 2019-06-29:
Now is ok, i can do OTA like in the example without problems but i have a problem when i want to do a http to see if i need to make OTA, for this i have a project on esp32 and esp8266 where i have two functions from below. On esp32 i pool every hour for version and subversion and if something is different i make OTA, for esp8266 i can't do that so i need to make a blind OTA or via a remote command.

**This down't work on esp8266 but works on esp32**
void check_ota()
{
	printf("Looking for a new firmware...\n");
	// configure the esp_http_client
	esp_http_client_config_t config = {
		.url = UPDATE_JSON_URL,
		.event_handler = _http_event_handler,
	};
	esp_http_client_handle_t client = esp_http_client_init(&config);
	// downloading the json file
	esp_err_t err = esp_http_client_perform(client);
	if(err == ESP_OK) 
	{
		uint16_t json_len = esp_http_client_read(client, rcv_buffer, sizeof(rcv_buffer));
		if(json_len > 0 && json_len < sizeof(rcv_buffer))
		{
			// parse the json file	
			cJSON *json = cJSON_Parse(rcv_buffer);
			if(json != NULL) 
			{
				cJSON *version = cJSON_GetObjectItemCaseSensitive(json, "version");
				cJSON *subversion = cJSON_GetObjectItemCaseSensitive(json, "subversion");
				cJSON *file = cJSON_GetObjectItemCaseSensitive(json, "file");
				printf("Compare %s with %s\n",version->valuestring, __DATE__);
				printf("Compare %s with %s\n",subversion->valuestring, __TIME__);
				if(strcmp(version->valuestring, __DATE__) || strcmp(subversion->valuestring, __TIME__))
				{
					printf("Downloading new firmware\n");
					if(cJSON_IsString(file) && (file->valuestring != NULL)) 
						make_ota(file->valuestring);	
					else printf("unable to read the new file name, aborting...\n");
				}
				else printf("Server send same firmware version\n");
			}
			else printf("downloaded file is not a valid json, aborting...\n");
		}
		else printf("downloaded file either empty or too big, size %d\n", json_len);
	}
	else printf("unable to download the json file, aborting...\n");
	// cleanup
	esp_http_client_cleanup(client);
	printf("\n");
}

**This works on both projects:**
void make_ota(char *url)
{
	printf("downloading and installing new firmware (%s)...\n", url);
	esp_http_client_config_t ota_client_config = {
		.url = url,
		.cert_pem = server_cert_pem_start,
		.event_handler = _http_event_handler,
	};
	esp_err_t ret = esp_https_ota(&ota_client_config);
	if (ret == ESP_OK) 
	{
		printf("OTA OK, restarting...\n");
		esp_restart();
	} 
	else printf("OTA failed...\n");
}

xpratpa, 2019-07-25:
I am getting the following error. Has anyone resolved this issue?  I have tried different INPUT length but it doesn't work for me. 
E (9686) esp-tls: read error :-29184:
E (9686) TRANS_SSL: esp_tls_conn_read error, errno=Success
D (9696) HTTP_CLIENT: need_read=256, byte_to_read=256, rlen=-29184, ridx=0
I (9706) esp_https_ota: Connection closed,all data received
E (9766) esp_https_ota: Error: esp_ota_end failed! err=0x258. Image is invalid
E (9776) simple_ota_example: Firmware Upgrades Failed

boriss2oo6, 2019-07-25:
@xpratpa it's solved for me now, i can successfully make OTA but i receive the same error as you posted if i open another http socket (and don't close it) before making OTA. If i try directly OTA like in the example provided in ..\examples\system\ota\simple_ota_example\main\simple_ota_example.c

xpratpa, 2019-07-26:
@boriss2oo6 Thanks for the quick response. It works for me if I set the INPUT length is 16384 and close all active connections as you mentioned before making OTA connection. It seems that heap memory is running out while making OTA on ESP8266 when more than one HTTPS or SSL connection is open.  It depends ...

donghengqaz, 2019-07-26:
Maybe you can talk more about the problem how ESP8266 does't work. And I can find more information about the difference of behavior from esp8266 and esp32.

boriss2oo6, 2019-07-26:
@donghengqaz I already described the problem in https://github.com/espressif/ESP8266_RTOS_SDK/issues/572#issuecomment-506930846

Basically I want to start a http connection to get the server's preferred firmware version and compare with running firmware version and make an OTA only if I need to. I use the __DATE__ and __TIME__ macros as a firmware version and subversion.

FayeY, 2019-08-22:
Sorry that I am a little confused, is this issue solved now? Does it due to out of memory?

FayeY, 2019-11-08:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details. Thanks.

closed 2019-11-08 by FayeY

nsfilho, 2020-05-20:
Hi everyone! I'm facing a problem like @boriss2oo6 in May/13, but my code is starting before 2MB address.

```txt
D (101206) esp_https_ota: Written image length 568480
D (101212) HTTP_CLIENT: is_data_remain=0, is_chunked=0
I (101220) esp_https_ota: Connection closed,all data received
D (101230) ota.c: HTTP_EVENT_DISCONNECTED
D (101241) ota.c: HTTP_EVENT_DISCONNECTED
D (101243) esp_https_ota: Total binary data length writen: 568480
D (101250) esp_image: reading image header @ 0xf1000
D (101257) esp_image: image header: 0xe9 0x05 0x02 0x02 40211bf0
I (101266) esp_image: segment 0: paddr=0x000f1010 vaddr=0x40210010 size=0x70540 (460096) map
0x40210010: _stext at ??:?

I (101351) esp_image: segment 1: paddr=0x00161558 vaddr=0x40280550 size=0x13e44 ( 81476) map
I (101367) esp_image: segment 2: paddr=0x001753a4 vaddr=0x3ffe8000 size=0x006b8 (  1720)
I (101370) esp_image: segment 3: paddr=0x00175a64 vaddr=0x40100000 size=0x00a50 (  2640)
I (101380) esp_image: segment 4: paddr=0x001764bc vaddr=0x40100a50 size=0x057c0 ( 22464)
D (101397) esp_ota_ops: OTA binary start entry 0x11be0, partition start from 0xf1000 to 0x1d2000

D (101407) esp_image: reading image header @ 0xf1000
D (101415) esp_image: image header: 0xe9 0x05 0x02 0x02 40211bf0
I (101424) esp_image: segment 0: paddr=0x000f1010 vaddr=0x40210010 size=0x70540 (460096) map
0x40210010: _stext at ??:?

I (101508) esp_image: segment 1: paddr=0x00161558 vaddr=0x40280550 size=0x13e44 ( 81476) map
I (101524) esp_image: segment 2: paddr=0x001753a4 vaddr=0x3ffe8000 size=0x006b8 (  1720)
I (101527) esp_image: segment 3: paddr=0x00175a64 vaddr=0x40100000 size=0x00a50 (  2640)
I (101538) esp_image: segment 4: paddr=0x001764bc vaddr=0x40100a50 size=0x057c0 ( 22464)
I (101628) esp_https_ota: esp_ota_set_boot_partition succeeded
I (101631) ota.c: Firmware Updated!
I (101738) wifi: state: 5 -> 0 (0)
I (101739) wifi: pm stop
D (101740) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:CASA16, ssid_len:6, bssid:0c:80:63:e0:09:a2, reason:8
D (101748) udp_sync: UDP sync close port 68
D (101754) tcpip_adapter: if0 start ip lost tmr: enter
D (101761) tcpip_adapter: if0 start ip lost tmr: interval=120
E (101769) wifi.c: STA: Disconnect reason 0x0008
I (101778) wifi.c: AP: SSID [e12aio_CC8C20], Password: [naoadiantatentar]
I (101789) wifi.c: STA: SSID [CASA16], Password: [8Gg43yAo72L4Vt2vzEttHnf8]
D (101801) nvs: nvs_set opmode 1 3
D (101807) nvs: nvs_set_blob ap.ssid 36
D (101813) nvs: nvs_set_blob ap.passwd 65
D (101819) nvs: nvs_set ap.authmode 1 4
D (101823) nvs: nvs_set ap.chan 1 1
D (101828) nvs: nvs_set ap.hidden 1 0
D (101834) nvs: nvs_set ap.max.conn 1 4
D (101839) nvs: nvs_set bcn.interval 2 100
D (101846) event: SYSTEM_EVENT_STA_STOP
D (101850) nvs: nvs_set_blob sta.apsw 2
D (101857) nvs: nvs_set_blob sta.pmk 32
D (101862) nvs: nvs_set_blob sta.apinfo 700
D (101871) nvs: nvs_set bssid.set 1 0
D (101873) nvs: nvs_set sta.scan_method 1 0
D (101879) nvs: nvs_set sta.sort_method 1 0
D (101885) nvs: nvs_set sta.lis_intval 2 3
D (101891) nvs: nvs_set sta.minrssi 1 -127
D (101897) nvs: nvs_set sta.minauth 1 0
D (101906) event: SYSTEM_EVENT_STA_START
I (101908) wifi.c: STA: trying connect
� (101915) event: SYSTEM_EVENT_AP_START
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 6116, room 16
tail 4
chksum 0x35
load 0x3ffe8408, len 24, room 4
tail 4
chksum 0xdc
load 0x3ffe8420, len 3456, room 4
tail 12
chksum 0x41
I (81) boot: ESP-IDF v3.3-rc1-80-gb74255d0 2nd stage bootloader
I (81) boot: compile time 13:00:14
I (82) qio_mode: Enabling default flash chip QIO
I (99) boot: SPI Speed      : 40MHz
I (111) boot: SPI Mode       : QIO
I (124) boot: SPI Flash Size : 4MB
I (136) boot: Partition Table:
I (148) boot: ## Label            Usage          Type ST Offset   Length
I (171) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (194) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (217) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (240) boot:  3 factory          factory app      00 00 00010000 000e1000
I (264) boot:  4 ota_0            OTA app          00 10 000f1000 000e1000
I (287) boot:  5 ota_1            OTA app          00 11 001d2000 000e1000
I (310) boot:  6 storage          Unknown data     01 82 002b3000 0007d000
I (333) boot:  7 nvs_key          Unknown data     01 04 00330000 00001000
I (356) boot: End of partition table
I (369) esp_image: segment 0: paddr=0x000f1010 vaddr=0x40210010 size=0x70540 (460096) map
0x40210010: _stext at ??:?

I (590) esp_image: segment 1: paddr=0x00161558 vaddr=0x40280550 size=0x13e44 ( 81476) map
I (625) esp_image: segment 2: paddr=0x001753a4 vaddr=0x3ffe8000 size=0x006b8 (  1720) load
I (626) esp_image: segment 3: paddr=0x00175a64 vaddr=0x40100000 size=0x00a50 (  2640) load
I (648) esp_image: segment 4: paddr=0x001764bc vaddr=0x40100a50 size=0x057c0 ( 22464) load
I (683) boot: Loaded app from partition at offset 0xf1000
E (694) boot: ERROR: app bin error, start_addr f1000 image_len 568448

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 6116, room 16
tail 4
chksum 0x35
load 0x3ffe8408, len 24, room 4
tail 4
chksum 0xdc
load 0x3ffe8420, len 3456, room 4
tail 12
chksum 0x41
I (45) boot: ESP-IDF v3.3-rc1-80-gb74255d0 2nd stage bootloader
```

As in above log:

- use of mbedtls: OK;
- download from http_client: OK;
- esp_https_ota: ESP_OK;
- no other tasks, except watchdog;

But after restart, shows me this strange error:
**E (694) boot: ERROR: app bin error, start_addr f1000 image_len 568448**

If anyone can help me with this, I will be very grateful.

My code is public, if anyone want to see:
[https://github.com/nsfilho/E12AIO3/blob/feat/ota/main/ota.c](https://github.com/nsfilho/E12AIO3/blob/feat/ota/main/ota.c)

Thank you in advance for any tips & tricks about how to solve that.

swetha-paladugu-oyo, 2020-06-29:
> @boriss2oo6 Thanks for the quick response. It works for me if I set the INPUT length is 16384 and close all active connections as you mentioned before making OTA connection. It seems that heap [...]

How did u figure out active connections and how did u close it... any api's available?

swetha-paladugu-oyo, 2020-07-01:
> @boriss2oo6 Thanks for the quick response. It works for me if I set the INPUT length is 16384 and close all active connections as you mentioned before making OTA connection. It seems that heap [...]

How did you close active connections.. please reply..

boriss2oo6, 2020-07-02:
Hi,

Every 1 minute i do a check_ota() and if i need to update i set a flag and remember the url for make_ota(). Then if i have the flag ota.need == true i call make_ota(ota.url)

void check_ota()
{
	printf("Looking for a new firmware...\n");
	// configure the esp_http_client
	esp_http_client_config_t config = {
		.url = UPDATE_JSON_URL,
		.event_handler = _http_event_handler,
	};
	esp_http_client_handle_t client = esp_http_client_init(&config);
	// downloading the json file
	esp_err_t err = esp_http_client_perform(client);
	if(err == ESP_OK) 
	{
		uint16_t json_len = esp_http_client_read(client, rcv_buffer, sizeof(rcv_buffer));
		if(json_len > 0 && json_len < sizeof(rcv_buffer))
		{
			// parse the json file	
			cJSON *json = cJSON_Parse(rcv_buffer);
			if(json != NULL) 
			{
				cJSON *version = cJSON_GetObjectItemCaseSensitive(json, "version");
				cJSON *subversion = cJSON_GetObjectItemCaseSensitive(json, "subversion");
				cJSON *file = cJSON_GetObjectItemCaseSensitive(json, "file");
				printf("Compare %s with %s\n",version->valuestring, __DATE__);
				printf("Compare %s with %s\n",subversion->valuestring, __TIME__);
				if(strcmp(version->valuestring, __DATE__) && strcmp(subversion->valuestring, __TIME__))
				{
					printf("Downloading new firmware\n");
					if(cJSON_IsString(file) && (file->valuestring != NULL))
					{
						ota.need = true;
						memcpy(ota.url, file->valuestring, strlen(file->valuestring));
					}
					else printf("unable to read the new file name, aborting...\n");
				}
				else printf("Server send same firmware version\n");
			}
			else printf("downloaded file is not a valid json, aborting...\n");
		}
		else printf("downloaded file either empty or too big, size %d\n", json_len);
	}
	else printf("unable to download the json file, aborting...\n");
	// cleanup
	esp_http_client_cleanup(client);
	printf("\n");
}

void make_ota(char *url)
{
	printf("downloading and installing new firmware (%s)...\n", url);
	esp_http_client_config_t ota_client_config = {
		.url = url,
		.cert_pem = server_cert_pem_start,	//openssl s_client -showcerts -connect hostmydream.com:443 < /dev/null
		.event_handler = _http_event_handler,
	};
	esp_err_t ret = esp_https_ota(&ota_client_config);
	if (ret == ESP_OK) 
	{
		printf("OTA OK, restarting...\n");
		esp_restart();
	} 
	else printf("OTA failed...\n");
}

swetha-paladugu-oyo, 2020-07-02:
@boriss2oo6 thanks for the quick reply.. but that wasn't my question.
I will tell my problem once again..
The example project downloads the complete ota file. It works..
But in my project when i call the same api's:

initialise_wifi();

ota_example();

only few kilo bytes of ota file gets downloaded and ota stops.

where ota_example is,
void ota_example()
{
    ESP_LOGI(TAG, "Starting OTA example...");

    /* Wait for the callback to set the CONNECTED_BIT in the
       event group.
    */
    xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT,
                        false, true, portMAX_DELAY);
    ESP_LOGI(TAG, "Connect to Wifi ! Start to Connect to Server....");

    ESP_LOGI(TAG, "URL:%s",TEST_FIRMWARE_UPGRADE_URL);

    esp_http_client_config_t config = {
        .url = TEST_FIRMWARE_UPGRADE_URL,
        .cert_pem = (char *)server_cert_pem_start,
        .event_handler = _http_event_handler,
    };
    esp_err_t ret = esp_https_ota(&config);
    if (ret == ESP_OK) {
        esp_restart();
    } else {
        ESP_LOGE(TAG, "Firmware Upgrades Failed");
    }
    while (1) {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

Fails with these logs:
E 70867) HTTP_CLIENT: need_read=2381, byte_to_read=512, rlen=512, ridx=1715
E (71057) HTTP_CLIENT: http_on_body 512
E (71157) ota: HTTP_EVENT_ON_DATA, len=512
E (71277) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (71417) HTTP_CLIENT: need_read=1869, byte_to_read=512, rlen=364, ridx=2227
E (71597) HTTP_CLIENT: http_on_body 364
E (71707) ota: HTTP_EVENT_ON_DATA, len=364
E (71817) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (77017) esp-tls: read error :-76:
E (77017) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
E (77027) HTTP_CLIENT: need_read=1505, byte_to_read=512, rlen=-1, ridx=2591
E (77217) esp_https_ota: Written image length 39455
E (77337) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (82477) HTTP_CLIENT: need_read=4096, byte_to_read=512, rlen=0, ridx=0
I (82477) esp_https_ota: Connection closed,all data received
E (82547) ota: HTTP_EVENT_DISCONNECTED
E (82677) ota: HTTP_EVENT_DISCONNECTED
E (82757) esp_https_ota: Total binary data length writen: 39455
D (82917) esp_image: reading image header @ 0x110000
D (83027) esp_image: image header: 0xe9 0x04 0x02 0x02 40220928
V (83157) esp_image: loading segment header 0 at offset 0x110008
V (83297) esp_image: segment data length 0x6f3ec data starts 0x110010
I (83447) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x6f3ec (455660) map
V (83767) esp_image: loading segment header 1 at offset 0x17f3fc
V (83797) esp_image: segment data length 0xffffffff data starts 0x17f404
E (83957) esp_image: unaligned segment length 0xffffffff
E (84097) esp_https_ota: Error: esp_ota_end failed! err=0x5379. Image is invalid
E (84287) ota: Firmware Upgrades Failed

Not sure why this is happening.. Can you please help.

swetha-paladugu-oyo, 2020-07-02:
I am calling ota_example() as a function instead of a task... Will that be a problem? For some reason i am not able to create task.. Watchdog gets triggered if a create a task.. So i have made it a function and increased CONFIG_MAIN_TASK_STACK_SIZE to 8192 instead of 3584.

swetha-paladugu-oyo, 2020-07-02:
Complete file got downloaded when i created a task.. and ota was successful..

amakukha, 2020-11-27:
> I am sorry that the application binary must located at place before offset "2MB", and I see you place it at "0x210000". Now the SDK does not support factory binary, so you can remove it and move the 2 OTAs partitions the previous place.

Hello. Are there plans to fix this and allow a factory binary after 2 MB, @donghengqaz ?