espressif/ESP8266_RTOS_SDK #575: nvs_flash_init depends on spiffs

maierkomor, 2019-05-02:
Hello,

I am building an application for an ESP8285 and need to reduce the application size so that I can use OTA. Therefore, I want to remove spiffs from the binary, But the application crashes, if I do not link against any spiffs function.

My build environment is make based running on master (2b579ef7). The host system is Linux, and the prototype target is a Wemos D1 Mini Lite with an ESP8285.

If my application has no reference to esp_spiffs_format it crashes during startup. Simply adding a reference to this function resolves this issue. I am not sure with what other functions it interacts.

The crash output is as follows:

�I (42) boot: ESP-IDF v3.2-dev-306-g2b579ef7 2nd stage bootloader
I (42) boot: compile time 10:17:01
I (42) boot: SPI Speed      : 40MHz
I (47) boot: SPI Mode       : DOUT
I (51) boot: SPI Flash Size : 1MB
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (80) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 factory          factory app      00 00 00010000 000c8000
I (95) boot:  4 coredump         Unknown data     01 03 000d8000 00008000
I (103) boot:  5 storage          Unknown data     01 82 000e0000 00008000
I (110) boot: End of partition table
I (115) boot: Defaulting to factory image
I (119) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x685f8 (427512) map
I (291) esp_image: segment 1: paddr=0x00078610 vaddr=0x3ffe8000 size=0x01abc (  6844) load
I (294) esp_image: segment 2: paddr=0x0007a0d4 vaddr=0x3ffe9abc size=0x001c0 (   448) load
I (298) esp_image: segment 3: paddr=0x0007a29c vaddr=0x40100000 size=0x06928 ( 26920) load
I (317) boot: Loaded app from partition at offset 0x10000
I (384) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE

Task stack [uiT] stack from [0x3fff0890] to [0x3fff168c], total [3584] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c

3fff1600  0x00000000 0x3fff0558 0x402104fc 0x4024ff10 0x00000000 0x3fff0558 0x402104fc 0x4021b763 0x28c70010 0x02008eb3 0x1700b000 0x00600194 0x8e940160 0x402128b3 0x00600194 0x00000010
3fff1640  0x00000001 0x00000000 0x00000105 0x4024fb00 0x00000007 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3ffe9c64 0x4021ba4f 0x00000000 0x00000000 0x3ffe9c64 0x4021ae17
3fff1680  0x00000000 0x00000000 0x00000000 0x00000000

        PC: 0x40252dba        PS: 0x00000030        A0: 0x4021b7a4        A1: 0x3fff1610
        A2: 0x40215b43        A3: 0x3fff163b        A4: 0x00000094        A5: 0x3fff1639
        A6: 0x40215aaf        A7: 0x00600194        A8: 0x3fff14f0        A9: 0x00000080
       A10: 0x00000061       A11: 0x000000ff       A12: 0x402104fc       A13: 0x3fff0558
       A14: 0x00000000       A15: 0x00000017       SAR: 0x00000002  EXCCAUSE: 0x00000009

The stack resolves to:
0x40252dba: /opt/ESP8266_RTOS_SDK/components/util/include/ibus_data.h:83
0x4024fb00: /opt/ESP8266_RTOS_SDK/components/nvs_flash/src/nvs_api.cpp:454
0x4021b763: /opt/ESP8266_RTOS_SDK/components/esp8266/source/system_api.c:166
0x4021ae17: /opt/ESP8266_RTOS_SDK/components/esp8266/source/startup.c:79

app_main is in one of the first ESP_LOGI at that time. The only functions called at that point is esp_chip_info. Referenced line 79 in app_main is not even compiled as it is #undef. The next function called is nvs_flash_init, which seems to be related to that topic, because if I uncomment this line, the crash does not happen...

Is there a way around this issue?

Thanks,
Thomas

supreetd21, 2019-05-03:
_Currently, NVS uses a portion of main flash memory through spi_flash_{read|write|erase} APIs._ 
So, along with not referencing any spi functions, you will have to make sure you do not include any nvs functions in your application code as well.

You can ascertain this behavior as your code doesn't crash when you comment nvs_flash_init function.

maierkomor, 2019-05-03:
Yes, the code does not crash, if I remove nvs_flash_init. But what can I do to have NVS but omit SPIFFS? Is there a way around this issue? I've tried linking against the spi_functions that NVS needs, but the crash still persists. What does SPIFFS do that NVS relies on?

donghengqaz, 2019-05-05:
I see it is belong to a unaligned access to the bus, could tell me "0x4021b7a4" is where at the code?

donghengqaz, 2019-05-06:
Thanks for your report, and the bug will be fixed.

maierkomor, 2019-05-06:
Thank you very much. Seems to be working on master now.

Regards,
Thomas

closed 2019-05-13 by donghengqaz