espressif/ESP8266_RTOS_SDK #577: Enabling ESP_AES breaks WPS

DKrepsky, 2019-05-07:
## Environment

- Development Kit: ESP12F
- IDF version: //34b51698
- Development Env: Make
- Operating System: Linux Mint
- Power Supply: external 3.3V

## Problem Description

Enabling ESP_AES breaks the build for WPS.

### Expected Behavior
libwps is compiled and linked.

### Actual Behavior
libwps have missing symbols.

### Steps to reproduce

1. Make sure ESP_AES is enabled at menuconfig `CONFIG_ESP_AES=y`.
2. Try to build ${IDF_PATH}/examples/wifi/wps.

### Code to reproduce this issue

Use the WPS example provided by the SDK.

## Debug Logs

```
/opt/esp8266/ESP8266_RTOS_SDK/components/esp8266/lib/libwps.a(wps_common.o):(.text.wps_decrypt_encr_settings+0x0): undefined reference to `aes_128_cbc_decrypt'
/opt/esp8266/ESP8266_RTOS_SDK/components/esp8266/lib/libwps.a(wps_common.o): In function `wps_decrypt_encr_settings':
(.text.wps_decrypt_encr_settings+0x5e): undefined reference to `aes_128_cbc_decrypt'
/opt/esp8266/ESP8266_RTOS_SDK/components/esp8266/lib/libwps.a(wps_attr_build.o):(.text.wps_build_encr_settings+0x4): undefined reference to `aes_128_cbc_encrypt'
/opt/esp8266/ESP8266_RTOS_SDK/components/esp8266/lib/libwps.a(wps_attr_build.o): In function `wps_build_encr_settings':
(.text.wps_build_encr_settings+0x88): undefined reference to `aes_128_cbc_encrypt'
collect2: error: ld returned 1 exit status

```

donghengqaz, 2019-05-08:
Thanks for your report, and I have added the functions.

DKrepsky, 2019-05-08:
It is fixed. Thanks!

closed 2019-05-08 by DKrepsky