espressif/ESP8266_RTOS_SDK #578: 打开 Disable ROM UART print 以后不会生成/build/下的project_template.bin

Werner42, 2019-05-08:
尝试了几次发现只要把打开 Disable ROM UART print这个选项打开，project_template.bin就不会自动生成。

donghengqaz, 2019-05-08:
我基于 master 测试没有这个问题，你不是改了什么？

Werner42, 2019-05-08:
没有更改，我现在重新pull了一下，更新到最新的master还是一样的问题。而且我是直接在example里比如get-started里测试的。

Werner42, 2019-05-08:
> 我基于 master 测试没有这个问题，你不是改了什么？

没有更改，我现在重新pull了一下，更新到最新的master还是一样的问题。而且我是直接在example里比如get-started里测试的。

Werner42, 2019-05-08:
```
-rw-rw-r--  1 hcg hcg  204656  5月  8 12:49 project_template.bin
-rw-rw-r--  1 hcg hcg   25024  5月  8 14:42 project_template.bin0x00000.bin
-rw-rw-r--  1 hcg hcg  201500  5月  8 14:42 project_template.bin0x10010.bin
-rwxrwxr-x  1 hcg hcg 1326472  5月  8 14:42 project_template.elf*
-rw-rw-r--  1 hcg hcg 1402634  5月  8 14:42 project_template.map
```
我刚才新打开Disable ROM UART print，然后make all以后可以看到，project_template.bin没有更新。依然是12：49分时关闭Disable ROM UART print选项时创建的。

Werner42, 2019-05-12:
我今天大概测试了几次，发现是打开Disable ROM UART print 以后生成的是project_template.bin0x00000.bin
-  project_template.bin0x10010.bin 两个文件。关闭以后生成的是project_template.bin。也就是说应该是打开Disable ROM UART print 以后make调用esptool.py elf2image的时候,没有使用选项 --version==2

donghengqaz, 2019-05-13:
这个应该是在同步版本的时候出现的问题，生成固件的参数错误，多谢你的支持，这个问题的修复已经提交了。

closed 2019-05-13 by Werner42