espressif/ESP8266_RTOS_SDK #579: newlib in normal mode crashes in putc

maierkomor, 2019-05-09:
With SDK version v3.2-dev-320-g3ba190e8, I am getting a crash in _putc_r after switching from nano newlib to the normal one.

I am not sure what is causing this. I wanted to switch to the normal newlib, to be able to use %f and %G formatting, but removing that code didn't change anything.

The crash is pointing to line 104 of newlib/libc/stdio/put.c

Is this a known issue? Is there a workaround?

Task stack [rtT] stack from [0x3fff2b94] to [0x3fff3390], total [2048] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c

3fff31c0  0x3ffefc50 0x00000005 0x01e00561 0x40100450 0x343a3e40 0x00002830 0x00000007 0x40284958 0x3ffefc26 0x00000007 0x00000000 0x3ffe8038 0x3fff3250 0x00000000 0x3fff3e14 0x4022d98a
0x40100450: vPortETSIntrUnlock at /opt/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:348

0x40284958: rom_i2c_writeReg at ??:?

0x4022d98a: xQueueGenericSend at /opt/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332

3fff3200  0x3fff51bc 0x7f038800 0x00000000 0x4028867b 0x00000000 0x00000000 0x00000050 0x3ffefc26 0x00000000 0x00000000 0x00000000 0x00000000 0x3ffe923c 0x00000000 0x3fff3250 0x40278329
0x4028867b: chip_v6_rxmax_ext_dig at ??:?

0x40278329: esp_event_send at /opt/ESP8266_RTOS_SDK/components/esp8266/source/event_loop.c:92

3fff3240  0x3fff51bc 0x3ffefb8c 0x3ffe9728 0x40227f16 0x00000005 0x69727461 0x00006d75 0x3fff1ab4 0x402672a3 0x0000000a 0x3ffef50c 0x40106ce8 0x40253305 0x00000006 0xc9000000 0x40286a7c
0x40227f16: cnx_start_handoff_cb at ??:?

0x402672a3: lock_release_generic at /opt/ESP8266_RTOS_SDK/components/newlib/newlib/port/locks.c:172

0x40253305: _putc_r at /home/dongheng/workspace/espressif/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/putc.c:104

0x40286a7c: chip_v6_set_chanfreq at ??:?

3fff3280  0x00000000 0x3fff51bc 0x3ffe92cf 0x3ffe9728 0x3ffef94c 0x3ffefb8c 0x3ffefc26 0x402276be 0x00000001 0xa5a5a5a5 0x00000000 0x40227e58 0x3fff51bc 0x3ffef82c 0x00000000 0x402259d7
0x402276be: chm_set_current_channel at ??:?

0x40227e58: cnx_start_handoff_cb at ??:?

0x402259d7: scan_build_chan_list at ??:?

3fff32c0  0x3fff3300 0x00000020 0x00000000 0x4021de61 0x3fff2588 0x3fff2b18 0xffffffff 0x3ffefad4 0x00000000 0x00000000 0x00000023 0x4021e044 0x00000002 0x00000000 0x3ffef82c 0x00000000
0x4021de61: wifi_timer_init at ??:?

0x4021e044: ets_timer_arm_ms_us at ??:?

----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at bbs.espressif.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:      [Arduino IDE|Make|Eclipse|other]
- Operating System:     [Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

donghengqaz, 2019-05-13:
I see the crash task is "rtT", and it is a real kernel task, I suggest that you can use the FreeRTOS timer instead of the "rtT".

maierkomor, 2019-05-13:
Unfortunately, my user code has no reference to esp timers. So I thought this might be related to run time stats, and switched from CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER to CONFIG_FREERTOS_RUN_TIME_STATS_USING_CPU_CLK. But it is still crashing, and the stack trace also reveals that wifi_timer_init references ets_timer_arm_ms_us. 

But I am not sure where this call is triggered. I found out that MDNS has references to esp_timer. But removing it didn't fix the crash and the rtT is still causing it. 

Do you have an idea what else might be referencing esp_timer? Are there plans to update MDNS to work without esp_timer?

The complete stack trace is as follows:
Task stack [rtT] stack from [0x3fff2b9c] to [0x3fff3398], total [2048] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34
        38         3c

3fff31c0  0x3ffefc58 0x00000005 0x01e00561 0x4010044c 0x343a3e40 0x00002830 0x00000007 0x4028498c 0x3ffefc2e 0x00000007 0x00000000 0x3ffe8038 0x3fff3250 0x00000000
0x3fff3e1c 0x4022d92e
0x4010044c: vPortETSIntrUnlock at /opt/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:348

0x4028498c: rom_i2c_writeReg at ??:?

0x4022d92e: xQueueGenericSend at /opt/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332

3fff3200  0x3fff51c4 0x7f038800 0x00000000 0x402886b3 0x00000000 0x00000000 0x00000050 0x3ffefc2e 0x00000000 0x00000000 0x00000000 0x00000000 0x3ffe9234 0x00000000
0x3fff3250 0x402782e1
0x402886b3: chip_v6_rxmax_ext_dig at ??:?

0x402782e1: esp_event_send at /opt/ESP8266_RTOS_SDK/components/esp8266/source/event_loop.c:92

3fff3240  0x3fff51c4 0x3ffefb94 0x3ffe9720 0x40227eba 0x00000005 0x69727461 0x00006d75 0x3fff1abc 0x40267267 0x0000000a 0x3ffef514 0x40106ce0 0x402532d5 0x00000006 0xc9000000 0x40286ab4
0x40227eba: cnx_start_handoff_cb at ??:?

0x40267267: lock_release_generic at /opt/ESP8266_RTOS_SDK/components/newlib/newlib/port/locks.c:172

0x402532d5: _putc_r at /home/dongheng/workspace/espressif/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/putc.c:104

0x40286ab4: chip_v6_set_chanfreq at ??:?

3fff3280  0x00000000 0x3fff51c4 0x3ffe92c7 0x3ffe9720 0x3ffef954 0x3ffefb94 0x3ffefc2e 0x40227662 0x00000001 0x401009c0 0x00000000 0x40227dfc 0x3fff51c4 0x3ffef834 0x00000000 0x4022597b
0x40227662: chm_set_current_channel at ??:?

0x401009c0: vListInsertEnd at /opt/ESP8266_RTOS_SDK/components/freertos/freertos/list.c:93

0x40227dfc: cnx_start_handoff_cb at ??:?

0x4022597b: scan_build_chan_list at ??:?

3fff32c0  0x3fff3300 0x00000020 0x00000000 0x4021de09 0x3fff2590 0x3fff2b20 0xffffffff 0x3ffefadc 0x00000000 0x00000000 0x00000023 0x4021dfec 0x00000002 0x00000000 0x3ffef834 0x00000000
0x4021de09: wifi_timer_init at ??:?

0x4021dfec: ets_timer_arm_ms_us at ??:?

3fff3300  0x00000000 0x3fff51c4 0x3ffef834 0x00000000 0x00000023 0x401068dc 0x3ffefabc 0x40225b51 0x00000000 0x00000b05 0x00000000 0x4022e6d8 0x00000000 0x3ffe9698 0x40225d24 0x00000000
0x401068dc: xthal_set_intset at ??:?

0x40225b51: scan_build_chan_list at ??:?

0x4022e6d8: xTaskResumeAll at /opt/ESP8266_RTOS_SDK/components/freertos/freertos/tasks.c:4455

0x40225d24: scan_start at ??:?

3fff3340  0x00000020 0x401068dc 0x3ffefadc 0x40225b70 0x00000000 0x00000000 0x40106834 0x4021e073 0x00000000 0x00000000 0x40106834 0x4021e0e4 0x40106890 0x00000000 0x00000000 0x00000000
0x401068dc: xthal_set_intset at ??:?

0x40225b70: scan_build_chan_list at ??:?

0x40106834: xthal_set_intset at ??:?

0x4021e073: ets_timer_handler_isr at ??:?

0x40106834: xthal_set_intset at ??:?

0x4021e0e4: ets_timer_handler_isr at ??:?
[... 74 lines cut ...]
0x40106834: xthal_set_intset at ??:?

0x4021de14: ets_timer_handler_isr at ??:?

0x40106890: xthal_set_intset at ??:?

3fff2c40  0x00000000 0x00000000 0x00000000

        PC: 0x401009de        PS: 0x00000033        A0: 0x4022e4e4        A1: 0x3fff2aa0
0x401009de: uxListRemove at /opt/ESP8266_RTOS_SDK/components/freertos/freertos/list.c:176

0x4022e4e4: xTaskRemoveFromEventList at /opt/ESP8266_RTOS_SDK/components/freertos/freertos/tasks.c:4455

        A2: 0x40105168        A3: 0x288019ec        A4: 0xec4e0100        A5: 0x0086ec4f
0x40105168: _xt_nmi at /opt/ESP8266_RTOS_SDK/components/freertos/port/esp8266/xtensa_vectors.S:755

        A6: 0xc9000000        A7: 0x40281a94        A8: 0x3fff2b30        A9: 0x00000000
0x40281a94: chip_v6_set_chanfreq at ??:?

       A10: 0x00000008       A11: 0x80000000       A12: 0x40105150       A13: 0x40105168
0x40105150: _xt_user_exit at /opt/ESP8266_RTOS_SDK/components/freertos/port/esp8266/xtensa_vectors.S:715

0x40105168: _xt_nmi at /opt/ESP8266_RTOS_SDK/components/freertos/port/esp8266/xtensa_vectors.S:755

       A14: 0x3fff2b00       A15: 0xfffffffe       SAR: 0x00000004  EXCCAUSE: 0x00000009

donghengqaz, 2019-05-17:
It is not an easy problem, and could you apply a base project which an cause the crash to me?

FayeY, 2019-06-27:
Is there any update about this issue?

maierkomor, 2019-06-30:
I just checked with idf v3.2-7-g73511106. Linking to the regular newlib doesn't cause a crash anymore for me. %f and %g also seem to be working now.

closed 2019-08-22 by FayeY