espressif/ESP8266_RTOS_SDK #581: 直接下载最新的代码，设置好环境编译无法通过

leovs, 2019-05-10:
make
WARNING: Toolchain version is not supported: 1.22.0-60-g37b07f6
Expected to see version: 1.22.0-92-g8facf4c
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Please download and use the toolchain from the URL of README.md
WARNING: Compiler version is not supported: 4.8.5
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Please download and use the toolchain from the URL of README.md
WARNING: Toolchain version is not supported: 1.22.0-60-g37b07f6
Expected to see version: 1.22.0-92-g8facf4c
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Please download and use the toolchain from the URL of README.md
WARNING: Compiler version is not supported: 4.8.5
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Please download and use the toolchain from the URL of README.md
CC build/esp8266/source/esp_err_to_name.o
/root/esp8266/ESP8266_RTOS_SDK/components/esp8266/source/esp_err_to_name.c:384:29: error: unknown type name 'esp_err_t'
 const char *esp_err_to_name(esp_err_t code)
                             ^
/root/esp8266/ESP8266_RTOS_SDK/components/esp8266/source/esp_err_to_name.c:399:31: error: unknown type name 'esp_err_t'
 const char *esp_err_to_name_r(esp_err_t code, char *buf, size_t buflen)
                               ^
/root/esp8266/ESP8266_RTOS_SDK/make/component_wrapper.mk:285: recipe for target 'source/esp_err_to_name.o' failed
make[1]: *** [source/esp_err_to_name.o] Error 1
/root/esp8266/ESP8266_RTOS_SDK/make/project.mk:469: recipe for target 'component-esp8266-build' failed
make: *** [component-esp8266-build] Error 2
root@leovs-desktop:~/esp8266/ESP8266_RTOS_SDK/examples/get-started/project_template#

donghengqaz, 2019-05-13:
Do you clone the "master" branch SDK ? And you should download the right cross toolchain in the README.md at the root directory.

judsonzhu, 2019-05-13:
我的也编译不过，系统Ubuntu16 64，log 如下
 make
make[1]: Entering directory '/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig'
cc -c  -D_GNU_SOURCE   -DCURSES_LOC="<ncurses.h>" -DLOCALE -MD  /home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/zconf.tab.c -o zconf.tab.o
In file included from /home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/zconf.tab.c:2449:0:
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c: In function ‘conf_get_default_confname’:
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c:113:23: error: ‘PATH_MAX’ undeclared (first use in this function)
  static char fullname[PATH_MAX+1];
                       ^
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c:113:23: note: each undeclared identifier is reported only once for each function it appears in
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c: In function ‘conf_write’:
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c:741:15: error: ‘PATH_MAX’ undeclared (first use in this function)
  char dirname[PATH_MAX+1], tmpname[PATH_MAX+1], newname[PATH_MAX+1];
               ^
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c: In function ‘conf_split_config’:
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c:837:12: error: ‘PATH_MAX’ undeclared (first use in this function)
  char path[PATH_MAX+1];
            ^
Makefile:171: recipe for target 'zconf.tab.o' failed
make[1]: *** [zconf.tab.o] Error 1
make[1]: Leaving directory '/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig'
make: *** No rule to make target '/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/conf', needed by '/home/zhuzt/work/esp/tbmqtt/ssl/build/include/config/auto.conf'.  Stop.
zhuzt@zhuzt-VirtualBox:~/work/esp/tbmqtt/ssl$ make
make[1]: Entering directory '/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig'
cc -c  -D_GNU_SOURCE   -DCURSES_LOC="<ncurses.h>" -DLOCALE -MD  /home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/zconf.tab.c -o zconf.tab.o
In file included from /home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/zconf.tab.c:2449:0:
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c: In function ‘conf_get_default_confname’:
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c:113:23: error: ‘PATH_MAX’ undeclared (first use in this function)
  static char fullname[PATH_MAX+1];
                       ^
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c:113:23: note: each undeclared identifier is reported only once for each function it appears in
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c: In function ‘conf_write’:
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c:741:15: error: ‘PATH_MAX’ undeclared (first use in this function)
  char dirname[PATH_MAX+1], tmpname[PATH_MAX+1], newname[PATH_MAX+1];
               ^
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c: In function ‘conf_split_config’:
/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig/confdata.c:837:12: error: ‘PATH_MAX’ undeclared (first use in this function)
  char path[PATH_MAX+1];
            ^
Makefile:171: recipe for target 'zconf.tab.o' failed
make[1]: *** [zconf.tab.o] Error 1
make[1]: Leaving directory '/home/zhuzt/work/esp/ESP8266_RTOS_SDK/tools/kconfig'

leovs, 2019-05-13:
我的知道原因了，是因为下载的xtensa-lx106-elf的版本不对，在linux下不可执行，后来又遇到和esp32环境冲突的问题，不过现在解决了。

你的这个看起来可能也是环境配置的问题，或许是版本不对。

donghengqaz, 2019-05-17:
建议按照 [文档](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/linux-setup.html) 搭建开发环境，不要漏了什么步骤。

wivin001, 2019-05-19:
你好楼主，你的问题解决了嘛？我这边遇到这个问题已经三天三夜了，还没有解决

leovs, 2019-05-25:
> 你好楼主，你的问题解决了嘛？我这边遇到这个问题已经三天三夜了，还没有解决

当天问完后就解决了，我是环境没有配置好。

wivin001, 2019-05-25:
> 
> 
> > 你好楼主，你的问题解决了嘛？我这边遇到这个问题已经三天三夜了，还没有解决
> 
> 当天问完后就解决了，我是环境没有配置好。

我也是解决了就是了 我被网上一些教程误解了，能留下联系方式 一起学习吗

closed 2019-06-27 by FayeY