espressif/ESP8266_RTOS_SDK #583: 部分模块联网停留再 REASON_AUTH_EXPIRE， REASON_ASSOC_EXPIRE 有时候有 REASON_ASSOC_LEAVE，请问怎么优化？

skylli, 2019-05-11:
目前 100pcs 里 15% 模块出现如下情况：

- smartconfig 往往非常快进入 `SC_STATUS_LINK` 获取到路由的 SSID、passwor,但是链接路由，做四次握手时经常提示 `disconnect`,  `reason=REASON_AUTH_EXPIRE` 或者 `reason=REASON_ASSOC_EXPIRE ` 过 5分钟 甚至更久就可以连接上了。 
- 直接  `wifi_station_set_config`,  `wifi_station_connect` 配置链接 AP 也会有这种情况。

是否可以初步断定问题出在 `esp8266` 链接 路由时出了问题， 请问这个问题如何解决？？
这些问题在网件 WR7600 路由上出现很频繁。
虽然给多点时间模块总能连接上，但是用户留给产品的时间不多，想知道这么去优化？？

FireWinter, 2019-05-28:
> 目前 100pcs 里 15% 模块出现如下情况：
>
> * `smartconfig`  往往非常快进入 `SC_STATUS_LINK` 获取到路由的  `SSID`、`passwor` ,但是链接路由，做四次握手时经常提示 `disconnect`,  `reason=REASON_AUTH_EXPIRE` 或者 `reason=REASON_ASSOC_EXPIRE ` 过 5分钟 [...]

请问你当前用的哪个sdk版本测的？这个问题应该已经fix掉了，建议用rtos的master版本测试看看

FayeY, 2019-06-27:
由于长时间无更新，此问题暂时关闭。如果最新的 SDK 仍然有问题，欢迎 reopen issue 继续讨论。

closed 2019-06-27 by FayeY

yangchao1p, 2020-12-27:
我在ESP8266 SDK 2.1.1 当中遇到了，请问这个问题有在2.1.1当中解决吗？如果没有的话，我自己有办法解决么？