espressif/ESP8266_RTOS_SDK #586: fatal error: mqtt_client.h: No such file or directory

munna1116, 2019-05-13:
Issue: mqtt_client.h No such file or directory found.

I have tried copying other examples and able to compile their was no issue but while trying this esp-mqtt example over tcp from **ESP8266_RTOS_SDK/examples/protocols/ esp-mqtt/tcp** I am facing this issue.  Please help me.  

Using ESP8266_RTOS_SDK

Thanks in advance

donghengqaz, 2019-05-14:
You should enable esp-mqtt at "menuconfig -> conponents config -> mqtt -> select esp_mqtt"。And we are fixing some default config lack.

hiperiondev, 2019-05-16:
There is no esp_mqtt in release/v3.1?

munna1116, 2019-05-16:
> You should enable esp-mqtt at "menuconfig -> conponents config -> mqtt -> select esp_mqtt"。And we are fixing some default config lack.

Thank you so much @donghengqaz , I have followed the steps that you explained now its working. 
Issue resolved. Thank you so much for the quick response.

donghengqaz, 2019-05-16:
@hiperiondev Yes, the master(release/v3.2) has esp-mqtt。

hiperiondev, 2019-05-16:
I'm little confused about branches. In README said:
"1. Master branch
The master branch is an integration branch where bug fixes/features are gathered for compiling and functional testing.
2. Release branch
The release branch is where releases are maintained and hot fixes (with names like release/v2.x.x) are added. Please ensure that all your production-related work are tracked with the release branches."

I I understand then master is like beta/probe release and not something for use in production.
Is master(release/v3.2) stable?

donghengqaz, 2019-05-17:
I mean that the release/v3.2 will publish and it is based on master, so "master" will create the  "release/v3.2" branch.

closed 2019-06-27 by FayeY