espressif/ESP8266_RTOS_SDK #587: How do I use esp_reset_reason()

iravan, 2019-05-14:
How do I use esp_reset_reason()  ? 
I have application that it needs reset reason. However , I found that esp_reset_reason() always return 0x02(ESP_RST_EXT) even the program was call esp_restart().

donghengqaz, 2019-05-16:
I tried the master branch and I see the esp_reset_reason() return 0x03。

donghengqaz, 2019-05-16:
Could you apply your base project to us?

iravan, 2019-05-17:
@donghengqaz 
```C
void app_main(void) {
 esp_reset_reason_t reason = esp_reset_reason();
 ESP_LOGI(tag , "Reset Reason : %X"  ,  reason);
 if(reason != ESP_RST_SW){
    esp_restart();
 }
 return;
}
```
That program reboots endlessly with log output.

> I (3418) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
> I (3608) reset_reason: RTC reset 2 wakeup 0 store 3, reason is 2
> I (3618) : Reset Reason : 2

iravan, 2019-05-23:
> @donghengqaz
> 
> ```c
> void app_main(void) {
>  esp_reset_reason_t reason = esp_reset_reason();
>  ESP_LOGI("" , "Reset Reason : %X"  ,  reason);
>  if(reason != ESP_RST_SW){
>     esp_restart();
>  }
>  return;
> }
> ```
> 
> That program reboots endlessly with log output.
> 
> > I (3418) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
> > I (3608) reset_reason: RTC reset 2 wakeup 0 store 3, reason is 2
> > I (3618) : Reset Reason : 2

I observe that it's about hardware environment on my side. My esp8266 module didn't have a stable power source. That might cause a reset before power up. So , the reset reason has no issue overall. 
Thanks.

closed 2019-05-23 by iravan