espressif/ESP8266_RTOS_SDK #595: download new version esp8266 release v3.1 and use idf compiler but it can't pass.

wivin001, 2019-05-19:
1、I use idf and change the windows toolchain new version from github ,when i compiler 
$  make all
Toolchain path: /opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-92-g8facf4c0
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-92-g8facf4c0
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
2、how can i download old version crosstool-ng-1.22.0-80-g6c4433a toolchain from esp8266 ？
3、or how can i solve this issue？

donghengqaz, 2019-05-21:
You can download the cross toolchain with URL in README.md.

wivin001, 2019-05-25:
> 
> 
> You can download the cross toolchain with URL in README.md.

sorry  已经解决了，我误解了网上的教程配置方式。按照开发文档来一定不会错的

closed 2019-06-24 by FayeY