espressif/ESP8266_RTOS_SDK #599: My application meet the error that "0x40100000: _stext at ??:?"

GzhuFlyer, 2019-05-23:
Hello, when I finished the instruction show in the readme.txt to build my develop environment ,and run the example that get-started/project_template. I get the message like this 

[image]

I (248) esp_image: segment 1: paddr=0x0003f3f8 vaddr=0x3ffe8000 size=0x0061c (  1564) load
I (249) esp_image: segment 2: paddr=0x0003fa1c vaddr=0x3ffe861c size=0x0011c (   284) load
I (259) esp_image: segment 3: paddr=0x0003fb40 vaddr=0x40100000 size=0x06900 ( 26880) load
0x40100000: _stext at ??:?

The main function like this 
//=======================================//
void app_main(void)
{
    printf("SDK version:%s\n", esp_get_idf_version());
}
//=======================================//
But the program not continue to run, I can't get the print message.
My PC system version :  Ubuntu 18.04.1 LTS,
the compile tools version : xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-92-g8facf4c) 5.2.0
the SDK version：release/v3.1

GzhuFlyer, 2019-05-23:
> Hello, when I finished the instruction show in the readme.txt to build my develop environment ,and run the example that get-started/project_template. I get the message like this
>
> [image]
>
> I (248) [...]

any suggestion will be appreciated.

hulonghui106, 2019-06-11:
#558

GzhuFlyer, 2019-06-11:
> #558

Thank your for you answer,I has been solve this question,I set the wrong Flash SPI mode, I change QIO to DIO ,then rebuild and download the program, the program is running .

closed 2019-06-24 by FayeY