espressif/ESP8266_RTOS_SDK #602: PWM问题

leovs, 2019-05-25:
学习pwm，想先测试一个呼吸灯的效果
通过官方的示例代码发现pwm好像没有效果
正常pwm应该电压是通过频率和占空控制有渐变效果
而实际的效果是 0 直接到 3.3  3.3又直接到0
不知道问题出现在哪里。

#define PWM_0_OUT_IO_NUM 12
#define PWM_1_OUT_IO_NUM 13
#define PWM_2_OUT_IO_NUM 14
#define PWM_3_OUT_IO_NUM 15

// PWM period 500us(2Khz), same as depth
#define PWM_PERIOD (500)

static const char *TAG = "pwm_example";

// pwm pin number
const uint32_t pin_num[4] = {
PWM_0_OUT_IO_NUM,
PWM_1_OUT_IO_NUM,
PWM_2_OUT_IO_NUM,
PWM_3_OUT_IO_NUM
};

// duties table, real_duty = duties[x]/PERIOD
uint32_t duties[4] = {
250, 250, 250, 250,
};

// phase table, delay = (phase[x]/360)*PERIOD
int16_t phase[4] = {
0, 0, 90, -90,
};

void app_main()
{
pwm_init(PWM_PERIOD, duties, 4, pin_num);
pwm_set_channel_invert(0x1 << 0);
pwm_set_phases(phase);
pwm_start();
int16_t count = 0;

while (1) {
    if (count == 20) {
        // channel0, 1 output hight level.
        // channel2, 3 output low level.
        pwm_stop(0x3);
        ESP_LOGI(TAG, "PWM stop\n");
    } else if (count == 30) {
        pwm_start();
        ESP_LOGI(TAG, "PWM re-start\n");
        count = 0;
    }

    count++;
    vTaskDelay(1000 / portTICK_RATE_MS);
}

}

官方代码， 不知道怎样能生成正旋波

xiongyumail, 2019-05-27:
@leovs 
使用`pwm_set_duty()`, `pwm_start()`接口来修改占空比。
需要达到渐变的效果，定时修改占空比即可。

xuhongv, 2019-05-29:
@leovs @xiongyumail 
这个是支持你们所说的 **平滑缓慢调光**，后续我会在我的仓库开源怎么使用，现在先给大家使用一下，使用环境是 rtos3.0 ：

[PWM.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3231126/PWM.zip)

closed 2019-06-24 by FayeY