espressif/ESP8266_RTOS_SDK #603: wps crashes on esp8266

maierkomor, 2019-05-25:
On master (v3.2-dev-353-gf49b40e7) and v3.1 (v3.1-51-g913a06a9) I get crashes during WPS handshake, while the same code runs on ESP32/v3.2-111-g524308c98 without a problem.

## Environment

- Development Kit:
- IDF version (see above)
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

Some seconds after starting wps pbc and triggering wps on the repeater, the system crashes.

### Expected Behavior

ip reception via wps

### Actual Behavior

crash

### Steps to repropduce

trigger wps and wait

### Code to reproduce this issue

The code is part of a bigger project. If necessary, I can share it privately. But I intend to make it publicly available later, so no NDA or so is necessary.

[crash-master.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/3219218/crash-master.txt)
[crash-v3.1.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/3219219/crash-v3.1.txt)
## Debug Logs
logs are attached as files

## Other items if possible

I tried to attach the elf and sdkconfig, but they are rejected as unsupported file types by the issue tracker.

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

FayeY, 2019-11-08:
Could you have a try with our latest WPS example？https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/wifi/wps
Please feel free to let us know if the issue persists with the latest version. 
Thanks.

maierkomor, 2020-04-07:
The issue still persists with the latest version (IDF version: v3.2-48-gb25b962d). I've attached the debug output of WiFi and WPS.
[wps.log](https://github.com/espressif/ESP8266_RTOS_SDK/files/4443290/wps.log)

maierkomor, 2020-04-07:
In contrast the example app seems to work. The log output gives:

I (269) boot: Loaded app from partition at offset 0x10000
I (308) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (308) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE

�I (378) phy_init: phy ver: 1058_13
I (378) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (388) example_wps: start wps...
I (398) example_wps: SYSTEM_EVENT_STA_START
I (9778) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
E (9778) example_wps: Disconnect reason : 23
I (10768) example_wps: SYSTEM_EVENT_STA_WPS_ER_SUCCESS
I (10778) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
E (10778) example_wps: Disconnect reason : 202
I (12968) event: sta ip: 192.168.11.215, mask: 255.255.255.0, gw: 192.168.11.1
I (12968) example_wps: SYSTEM_EVENT_STA_GOT_IP
I (12968) example_wps: got ip:192.168.11.215

maierkomor, 2020-04-07:
It turned out that this is an out-of-memory situation. So WPS is simply asking for too much RAM. After turning off every software service of Atrium, WPS completed successfully.

Can someone work on reducing the RAM requirements of WPS? I will try to update the services in Atrium, so that they only start if Station mode is up.

Additionally, it would be nice if the out-of-memory situation would not trigger a SegV, but return an error-code.

Thanks!