espressif/ESP8266_RTOS_SDK #604: Toolchain setup issue ?

dsptech, 2019-05-25:
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ X] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment
Not an hardware issue

## Problem Description
I'm a newbe about the ESP8266 RTOS SDK.
I'm try to install the environment for the first time (in the ESP32 fashion) but I aborted the installation because tool/windows/windows_install_prerequisites.sh try to download and install the ESP32 toolchain. 

I would like to know if there is an error in this file, or if the file must be ignored.
Thank you.
Regards.

`
........
# Automatically download precompiled toolchain, unpack at /opt/xtensa-esp32-elf/
TOOLCHAIN_ZIP=xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
echo "Downloading precompiled toolchain ${TOOLCHAIN_ZIP}..."
cd ~
curl -LO --retry 10 http://dl.espressif.com/dl/${TOOLCHAIN_ZIP}
cd /opt
rm -rf /opt/xtensa-esp32-elf  # for upgrades
unzip ~/${TOOLCHAIN_ZIP}
rm ~/${TOOLCHAIN_ZIP}

cat > /etc/profile.d/esp32_toolchain.sh << EOF
# This file was created by ESP-IDF windows_install_prerequisites.sh
# and will be overwritten if that script is run again.
export PATH="\$PATH:/opt/xtensa-esp32-elf/bin"
EOF
....
`

donghengqaz, 2019-05-27:
You should download the ESP8266 cross toolchain, the install document is not supported. And I reggust that  you can use ubuntu to develop, becasue it is much ealier.

dsptech, 2019-05-27:
Hi @donghengqaz,
I'm already download and installed that toochain that work correctly on mingw32.
I also configured Eclipse CDT for windows. It work correctly and also my fixes (for built-in settings provider) that I written for ESP32 work fine also with this framework.

I opened the issue because I was a bit concerned about the behavior of windows_install_prerequisites.sh, 
... and now you said me that the file is not supported, so, I will ignore the file.
Thank you.

donghengqaz, 2019-05-30:
I am sorry to say, because only very few people to develop the SDK, so fix or feature is published so later. Because of this, I give you that suggustion.

And we have been trying to fix the windows install scripts.

dsptech, 2019-06-09:
Hi @donghengqaz ,
I updated the framework yesterday, so I now able to test again the script.
windows_install_prerequisites.sh now work fine.

There are still some errors when extracting the toolchain (symbolic links) but they are toolchain related only because the same errors occur also when the file is extracted by hand. I think that the operation require the administrative rights.

Thank you.
Regards.

closed 2019-06-09 by dsptech