espressif/ESP8266_RTOS_SDK #605: I2C总线时序错误

zgh551, 2019-05-26:
## Environment

- Development Kit:Node Mcu
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): V3.1
- Development Env:      VSCode
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

目前使用ESP8266的I2C总线去读取传感器的数据，当把传感器从BOOT模式切换到APP模式后，总线时序就开始出现异常。
如下图所示，我实际想读取的寄存器地址为0x20，但是实际总线时序错乱后，发的却是0x41，同时ACK和Stop位也都错位了。

[image]

下面两张图是分别取读取0x00地址数据的波形图。
第一种情况寄存器地址0x00，直接发送错误；
[image]
第二种情况，从机返回了数据0x90，但是STOP位貌似错误了。
[image]

### Code to reproduce this issue

- I2C配置
```cpp
esp_err_t i2c_master_init()
{
    int i2c_master_port = I2C_EXAMPLE_MASTER_NUM;
    i2c_config_t conf;
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = I2C_EXAMPLE_MASTER_SDA_IO;
    conf.sda_pullup_en = 0;
    conf.scl_io_num = I2C_EXAMPLE_MASTER_SCL_IO;
    conf.scl_pullup_en = 0;
    ESP_ERROR_CHECK(i2c_driver_install(i2c_master_port, conf.mode));
    ESP_ERROR_CHECK(i2c_param_config(i2c_master_port, &conf));
    return ESP_OK;
}
```
- 写数据
```cpp
esp_err_t i2c_master_write_data(i2c_port_t i2c_num, uint8_t sensor_add, uint8_t reg_address, uint8_t *data, size_t data_len)
{
    int ret;
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, sensor_add << 1 | WRITE_BIT, ACK_CHECK_EN);
    i2c_master_write_byte(cmd, reg_address, ACK_CHECK_EN);
    i2c_master_write(cmd, data, data_len, ACK_CHECK_EN);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    return ret;
}
```

- 读数据
```cpp
esp_err_t i2c_master_read_data(i2c_port_t i2c_num, uint8_t sensor_add, uint8_t reg_address, uint8_t *data, size_t data_len)
{
    int ret;

    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, sensor_add << 1 | WRITE_BIT, ACK_CHECK_EN);
    i2c_master_write_byte(cmd, reg_address, ACK_CHECK_EN);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);

    if (ret != ESP_OK) {
        return ret;
    }

    cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, sensor_add << 1 | READ_BIT, ACK_CHECK_EN);
    i2c_master_read(cmd, data, data_len, LAST_NACK_VAL);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);

    return ret;
}
```

上述代码就是用的官方的例程。

## 思考

- 推测1
是否有可能从机进入APP模式后干扰了I2C总线，导致后续的时序错位。但是我使用CP2112这款USB转IIC的模块去读取传感器，总线是正常的，不管传感器是否进入APP模式。
- 推测2
因为ESP8266是软件模拟实现的IIC总线，不是专用的IIC外设，会不会在某些操作下影响了系统延时，导致总线出错。从源码上看，时序部分用了很多延时函数。
- 推测3
IIC总线频率问题，导致的总线时序不稳定。总线时钟算了一下，大概是100k，请教一下在哪里可以修改该频率。我可以降频试试。

xiongyumail, 2019-05-27:
@zgh551 

你可以尝试用下面这个补丁，增加了clock stretch功能。

另外可以修改i2c.c中的`i2c_master_wait`宏来改变频率。

```
git apply i2c_stretch.patch
```

[i2c_stretch.tar.gz](https://github.com/espressif/ESP8266_RTOS_SDK/files/3221631/i2c_stretch.tar.gz)

zgh551, 2019-05-27:
- 您好，尝试了上述的补丁，结果输出以下乱码`*P*VETTKuU+UZZTTTQTQQjP,uZT(UEQZUEuTQQjP+Eku`，也尝试修改不同的Stretch值，但是始终无法正常工作。

- 我看源码中`i2c_master_wait`对应的`os_delay_us`函数，请问没有一个可以全局修改的参数嘛，不然我看需要修改很多地方。

zgh551, 2019-05-27:
另外在读取错误的时候，会出现一下程序跑飞的情况
```
I (715) CCS811: Hardware ID:0x81
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff at 0x4023f59e
0x4023f59e: ccs811_init at C:/msys32/home/zgh551/esp/IOT/components/sensor/ccs811/src/ccs811.c:12 (discriminator 9)

file: "ccs811.c" line 11
func: ccs811_init
expression: i2c_master_read_data(i2c_num,CCS811_SENSOR_ADDR,HW_VERSION,i2c_buff,2)
E (735) ABORT: Error found and abort!

Task stack [ccs811_task] stack from [3fff188c] to [3fff288c], total [4096] size

          00000000 00000004 00000008 0000000c 00000010 00000014 00000018 0000001c 00000020 00000024 00000028 0000002c 00000030 00000034 00000038 0000003c
```

zgh551, 2019-05-27:
那个补丁修复后，读取错误出现如下乱码
```
I (584) flash_nvs: NVS Flash Init
I (594) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
I (604) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
I (714) CCS811: Hardware ID:0x81
I (714) CCS811: Hardware Version:0x12
I (724) CCS811: Fireware Boot Version:0x10
I (734) CCS811: Fireware App Version:0x20
I (734) CCS811: Status:0x10
I (734) CCS811: Valid application firmware loaded
I (744) CCS811: Write Status 0
I (744) CCS811: Write Status -1
I (754) CCS811: Receive Data 0x0 0x0 0x0 0x0
*P*VETTKuU+UZZTTEuTQQjP,uZT(UEQHPqU*BBuE+T)UZVUEuQBUUuZTUQTj*QZUEj*QZUUUPRBuU**PUZUuuUZUEUUQUUP
QUUTUUPRBuUTUUTUTUUTUUT*uUTU+UPIUUUTUUT*uUTZUUUJUTUUTU*uQ(*RUZUEUUQUZ
T(UP.U]UZJPUVUEUUQUZ-RQQQQQEUTUUT+UZ
U)Q
QQUQQQUTUUTUBuURZRZE,PQQQQQQQEQUEUUQTu+PKZTTUUTUUTTBXVVUZQTQ(QTQTQTUTQU\UTE(kUU(JUUUP*U-UVVUZTQ(QUTTQQU\Q)Q(TBTUEUUQUUU-UVVUZUTQ(QURTQU\Q*Q(BTUEUUQUUU-UVVUZUTQ(QU(TQTQ
QQBUTUEUUQUUUZTJPXUT]UZJPqQQU
```

xiongyumail, 2019-05-28:
@zgh551 
外部有没有上拉？

zgh551, 2019-05-28:
有的，硬件上有加上拉电阻

xiongyumail, 2019-05-28:
我这没复现出来，在IO13, IO14上挂了3个I2C设备，没有异常情况。

xiongyumail, 2019-05-28:
把设备断开，再看下波形？

zgh551, 2019-05-28:
你好，设备断开后，执行到i2c读取函数后，程序就奔溃了

zgh551, 2019-05-28:
- 我在I2C总线上挂载了两个设备，其中一个能读取出设备ID，另一个读取设备ID程序就奔溃了。
- 有个传感器有两种工作模式，上电默认工作在boot模式，在该模式下，读取设备ID，Version and Status 信息都正常，但是我将它配置到APP模式后，就不能正常读取status信息了，只能读取传感器值的信息。但是因为没有配置启动转换，所以读取的值一直不对。但是配置传感器命令，在进入APP模式后，执行I2C写返回值为是-1，也就是写入不成功

zgh551, 2019-05-28:
- 但是我使用CP2112这款USB转I2C接口去操作，都是正常的。不管传感器是否进入APP还是在BOOT模式下，都是可以正常操作。

xiongyumail, 2019-05-28:
在程序里把所有assert先去掉，我们首先要保证波形是正确的。从设备可能控制了SCL或SDA导致时序不正确。

zgh551, 2019-05-31:
- 如下图所示，红色框中时钟丢失
[image]

- 修改驱动引脚的输出模式，修改内容如下:
[image]

- 引脚输出模式修改后，输出波形如下，clk前几个时钟周期有了，但是幅值偏小。
[image]

- 但是输出模式修改后，IIC读取就不正常了

redchenjs, 2019-05-31:
I2C的GPIO必须设置为开漏输出，不能设置成推挽输出，否则强上拉可能将电流灌入从设备造成IO口损坏，另外看到信号波形中噪声很明显，建议检查上拉电阻及示波器探头地端是否良好接触。

zgh551, 2019-05-31:
如果配置成开漏输出，出现的clk信号丢失该怎么解决了

redchenjs, 2019-05-31:
clk信号丢失的问题并不是强行输出信号能解决的，图中出现的信号幅度低的现象恰恰说明了在I2C主机强行拉高IO口的同时从机正在输出低电平阻止clk信号的跳变，这样的信号仍然是无效的。
从机拉低clk信号的原因应该数据手册中有说明，比如正在准备数据、内部逻辑需要、未收到ACK位等等。
楼主提到硬件I2C能够工作的一个可能原因是硬件I2C有错误检测机制，会等待从机释放时钟线后才发送数据，不会漏发，软件I2C缺少这种机制，只顾一股脑地发送波形，不管时钟线是否空闲，造成实际波形缺失，通信失败。
根据波形图中时钟信号的缺失情况在通信过程中加入适量延时，应该能够避开此问题。

metadata1984, 2019-06-07:
部分从设备处理数据时就会尝试拉低时钟，让主设备暂缓发送数据。
此问题可能从设备硬件下拉不足，主设备不理会继续发时钟，就会出现半高的波形。
此问题软件i2c同样能解决，主设备拉低时钟后，可以轮询检查引脚是否被真的拉低。只有真的拉低后再发送数据。当然轮询需要设置超时时间。

redchenjs, 2019-06-07:
半高波形是主机IO被配置成推挽输出造成的，总线无法被拉低，正常情况使用开漏输出不会出现这种波形，楼主原先所说的clk丢失正是从机拉低clk的表现

closed 2019-06-24 by FayeY