espressif/ESP8266_RTOS_SDK #607: wifi路由器断电后，已经联网的ESP8266出错重启。

huozhe0930, 2019-05-27:
SDK版本，当前最新RTOS SDK3.2

测试了两次，每次都会出错重启（按打印信息显示来看，出错模块应该是esp_event_loop）
在wifi断开后，调用xEventGroupClearBits清除了连接的事件标志。
相关代码片段：（wifi的 event handler）
    case SYSTEM_EVENT_STA_GOT_IP:
		ESP_LOGI(TAG, "!!connect is established!\n");
		
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
        sntpStart( );
        if(xTcpClientHandle == NULL)
            xTaskCreate(tcp_client_task, "tcp_client", 4096, NULL, 5, &xTcpClientHandle);
        
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
	    ESP_LOGI(TAG, "!!connect is lost!\n");
		if(reconnectCounter++>50) {
			esp_wifi_stop( );
			esp_wifi_restore( );
		    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
			reconnectCounter = 0;
		    esp_wifi_start( );
		    break;
		}
		else
		{
			esp_wifi_connect();
            
		}
        if(xTcpClientHandle != NULL)
            vTaskDelete(&xTcpClientHandle);
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
重启前后打印信息如下：
[鞛*妀786236) wifi: err2:800312[0m
[0;32mI (806716) main: !!connect is lost!
[0m

Task stack [esp_event_loop_] stack from [0x3fff4a50] to [0x3fff524c], total [2048] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c 

3fff5100  0x000689d0 0x00000000 0x00000000 0x00000003 0x000003f0 0x40211bbb 0x00000002 0x00000020 0x3ffee3b8 0x3ffe8008 0x00000000 0x0000001f 0x00000000 0x00000000 0x00000000 0x40239cd2 
3fff5140  0x00000000 0x00000000 0x00000100 0x00000000 0x00000000 0x00000000 0x3ffef16c 0x40211698 0x40106fc0 0x00000000 0x40106fc0 0x4023f3ac 0x00000000 0x0001d4c0 0x00000000 0x4022e098 
3fff5180  0x00000000 0x3fff51a0 0x40106fc0 0x4022b840 0x00000000 0x40106fbc 0x40107128 0x40231128 0x00000000 0x4023f66c 0x3fff51c0 0x4022cdf0 0x40211698 0x40211698 0x00000003 0x00000000 
3fff51c0  0x00000000 0x40105fb8 0x00000000 0x4023f48e 0x00000000 0x40105fb8 0x00000000 0x4023fb50 0xffffffff 0x00000000 0x3fff5210 0x00000000 0x00000000 0x00000000 0x00000000 0x40247bd4 
3fff5200  0x00000000 0x00000000 0x00000000 0x40247c12 0x00000005 0x646f7568 0x6e6f7961 0x00000067 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x29b4f009 0xc825ed50 0x4022c62e 

3fff5240  0x00000000 0x00000000 0x00000000 0x00000000 

        PC: 0x40100a3e        PS: 0x00000033        A0: 0x4022b5be        A1: 0x3fff5130
        A2: 0x3ffedda0        A3: 0x00000000        A4: 0x3fff5eb8        A5: 0x3fff39b8
        A6: 0x00000000        A7: 0x00000000        A8: 0x00000003        A9: 0x000003f0
       A10: 0x40211bbb       A11: 0x00000002       A12: 0x3ffedd9c       A13: 0x3ffedda0
       A14: 0x00000000       A15: 0x00000000       SAR: 0x0000001f  EXCCAUSE: 0x0000001c

曼
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7188, room 16 
tail 4
chksum 0x88
load 0x3ffe8408, len 24, room 4 
tail 4
chksum 0x61
load 0x3ffe8420, len 3556, room 4 
tail 0
chksum 0x4a
csum 0x4a
[0;32mI (43) boot: ESP-IDF v3.2-dev-353-gf49b40e7-dirty 2nd stage bootloader[0m
[0;32mI (43) boot: compile time 14:20:42[0m
[0;32mI (44) boot: SPI Speed      : 40MHz[0m
[0;32mI (51) boot: SPI Mode       : DIO[0m
[0;32mI (57) boot: SPI Flash Size : 4MB[0m
[0;32mI (63) boot: Partition Table:[0m
[0;32mI (68) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (80) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (91) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (103) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (114) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (126) boot:  4 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (137) boot: End of partition table[0m
[0;32mI (144) boot: No factory image, trying OTA 0[0m
[0;32mI (152) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x536d0 (341712) map[0m
[0;32mI (308) esp_image: segment 1: paddr=0x000636e8 vaddr=0x3ffe8000 size=0x00744 (  1860) load[0m
[0;32mI (309) esp_image: segment 2: paddr=0x00063e34 vaddr=0x3ffe8744 size=0x0019c (   412) load[0m
[0;32mI (319) esp_image: segment 3: paddr=0x00063fd8 vaddr=0x40100000 size=0x063a4 ( 25508) load[0m
[0;32mI (343) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (386) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (396) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (576) phy_init: phy ver: 1055_12[0m
[0;32mI (586) reset_reason: RTC reset 2 wakeup 0 store 3, reason is 2[0m
[0;32mI (596) main: ==esp8266 SDK version:[v3.2-dev-353-gf49b40e7-dirty]==

donghengqaz, 2019-05-31:
`vTaskDelete(&xTcpClientHandle);` 这个不对吧？

huozhe0930, 2019-05-31:
您好，vTaskDelete(&xTcpClientHandle); 好像是我的问题，不应该带取址符的，谢谢。我再试试。

closed 2019-05-31 by huozhe0930