espressif/ESP8266_RTOS_SDK #609: how to set thread prios correctly?

dsyleixa, 2019-05-28:
forwarded from this discussion, 
https://github.com/espressif/esp-idf/issues/3544#issuecomment-496157019
there are still  issues about how to set thread prios correcty.

[CODE]
esp_pthread_cfg_t cfg;
  esp_pthread_get_cfg(&cfg);
  cfg.prio=1;
  esp_pthread_set_cfg(&cfg);[/CODE]

is assumed not to work because cfg might be empty and the function may return an error
and

[CODE]
  esp_pthread_cfg_t cfg;
  if (esp_pthread_get_cfg(&cfg) != ESP_OK) {
      cfg = esp_pthread_get_default_config();
  }
  cfg.prio=1;
  if (esp_pthread_set_cfg(&cfg) != ESP_OK) {
      printf("esp_pthread_set_cfg failed\n");
      abort();
  };[/CODE]

does not work, because 
esp_pthread_get_default_config();
is not found by the compiler.

Which is the coreect way, or what am I missing?

dsyleixa, 2019-05-29:
after having fixed that, the wishful would be for the future, e.g. like for POSIX pthread

    param.sched_priority = 1;
    pthread_setschedparam(thread0, SCHED_RR, &param);
   
    param.sched_priority = 2;
    pthread_setschedparam(thread1, SCHED_RR, &param);
   
        param.sched_priority = 5;
    pthread_setschedparam(thread2, SCHED_RR, &param);

donghengqaz, 2019-05-30:
Is your problem is about esp-idf(ESP32)？If so, the issue should be moved to esp-idf issues place.

dsyleixa, 2019-05-30:
yes, but it has been stated that it's about issues in the pthread libs.
Additionally it turned out that there are watchdog and scheduler issues in the pthread/std::thread implementation about preemptivity.

me21, 2019-07-29:
You're posting an issue for ESP32 in ESP8266 repo.

dsyleixa, 2019-07-29:
sorry, I missed that - where is the RTOS repo for ESP32?

me21, 2019-07-29:
https://github.com/espressif/arduino-esp32 you've already opened an issue there.

me21, 2019-07-29:
https://github.com/espressif/esp-idf

dsyleixa, 2019-07-29:
well - that is not actually about the underlying ESP32 RTOS, is it?

me21, 2019-07-29:
It is.

dsyleixa, 2019-07-29:
ok, thank you!

closed 2019-08-22 by FayeY