espressif/ESP8266_RTOS_SDK #611: pwm设置相位时候，发送：phase error 180, valid ramge from (-180,180)

xuhongv, 2019-05-29:
## Environment

- Development Kit:
- IDF version： v3.2-dev-342-g02382a2c
- Development Env:      [Make]
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

  最近在做pwm互补的驱动工程，看到新版的sdk支持相位设置，但是设置相位时候，设置 180时候， 提示错误！难道范围是 -179 到 179 ? 

### 设置179时候，波形如下，提示：` phase error 180, valid ramge from (-180,180)`：

[image]

## Other items if possible

- [ ] 现在我修改底层驱动把范围调整为 大于181 ，现在没事了！

[image]

- [ ] 波形截图.
[image]

xiongyumail, 2019-05-30:
@xuhongv 

需要反相可以用`pwm_set_channel_invert()`接口，
不过驱动里也可以改成[-180, 180]

xuhongv, 2019-05-31:
感谢！

closed 2019-05-31 by xuhongv