espressif/ESP8266_RTOS_SDK #613: CMake doesn't compile

paoloach, 2019-05-30:
## Environment
- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):  22d8fc64
- Development Env:      [other]
- Operating System:     [Ubuntu]
- Power Supply:         [USB]

## Problem Description

The CMake doesn't produce a right makefile.

### Expected Behavior

the makefile produced by CMake is usable
### Actual Behavior
the makefile produced by CMake compile untile the 56%. After I got the error:
/home/paolo/esp2866/espressif/ESP8266_RTOS_SDK/components/esp-tls/esp_tls.c:23:25: fatal error: http_parser.h: No such file or directory

### Steps to repropduce

1. go into examples/get-started/project_template
2. mkdir buid
3. cd build
4 cmake ../
5 make

### Code to reproduce this issue
- [ ] sdkconfig file is the default one produced by CMake

donghengqaz, 2019-05-31:
Thanks for your report.

Recently I have update components, and forgot updating some cmake configuration. I will fix this as soon as possible.

FayeY, 2019-07-15:
cmake is supported in the latest v3.2, please have a try. Thanks.

closed 2019-07-15 by FayeY