espressif/ESP8266_RTOS_SDK #614: i2c请教

leovs, 2019-05-30:
有个温度传感器，用的i2c的方式
用Arduino的环境好使，用rtos sdk一直调试不通

用调试器查看通信时序
在发送到0xb8后有个ack
然后后续的 3， 2， 2 的消息都没有发送。

cmd = i2c_cmd_link_create();
i2c_master_start(cmd);
i2c_master_write_byte(cmd, 0xb8, ACK_CHECK_EN);
i2c_master_write_byte(cmd, 0x03, ACK_CHECK_EN);
 i2c_master_write_byte(cmd, 0x02, ACK_CHECK_EN);
i2c_master_write_byte(cmd, 0x02, ACK_CHECK_EN);
i2c_master_stop(cmd);
i2c_master_cmd_begin(I2C_MASTER_NUM, cmd, 1000 / portTICK_RATE_MS);
i2c_cmd_link_delete(cmd);

用i2c_master_write 也不行。

xiongyumail, 2019-05-30:
@leovs 
请问是什么传感器型号呢?

leovs, 2019-05-30:
AM2320  温度传感器

leovs, 2019-05-30:
我觉着可能是我的代码有问题
[image]
以上的部分只看到了
i2c_master_write_byte(cmd, 0xb8, ACK_CHECK_EN);
的执行
后续的 3， 2 ，2  没看到

kmalloc8, 2019-06-19:
请问现在这个问题咋样了？

FayeY, 2019-06-24:
由于长时间无更新，此问题暂时关闭，如果问题仍未解决，欢迎 reopen issue 继续讨论。
这里有一个相似问题，#605，希望能有参考作用。

closed 2019-06-24 by FayeY