espressif/ESP8266_RTOS_SDK #619: SYSTEM_EVENT_STA_GOT_IP :  ip_changed property.

dsptech, 2019-06-06:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at bbs.espressif.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

commit: master branch 25/May/2019
----------------------------- Delete above -----------------------------

## Environment

- Development Kit: NodeMcu 1.0
- IDF version commit: master branch 25/May/2019
- Development Env:      [Eclipse]
- Operating System:     [Windows|
- Power Supply:         [USB]

## Problem Description

reason: ip_changed in SYSTEM_EVENT_STA_GOT_IP event is always true (in station mode)

### Expected Behavior
ip_changed = false for new ip
ip_changed = true for ip changes by DHCP

### Actual Behavior
ip_changed always true

### Code to reproduce this issue
use "simple_wifi" example and print  the ip_changed  property in SYSTEM_EVENT_STA_GOT_IP event

Please note: I found a similar post in espressif/esp-idf#2372 issue (https://github.com/espressif/esp-idf/issues/2372#issuecomment-419062809), so, does the ip_changed property work as intended ? 
(aka: what is the condition to obtain ip_changed = false ?)

donghengqaz, 2019-06-12:
When system starts, the default IP address is "0", so after getting real IP address from AP, the value must be "ip_changed". But if the Wi-Fi connection is lost, and then gets it after just a moment, perhaps the IP address is not changed, then "ip_changed = false".

dsptech, 2019-06-12:
Now it's clear.
Thank you.

closed 2019-06-15 by dsptech

burkulesomesh43, 2020-07-27:
Hi @dsptech,

How you handled if IP is changed?
needs to clear all socket connections.