espressif/ESP8266_RTOS_SDK #622: examples OTA问题

wzbhbb, 2019-06-10:
## Environment

- Development Kit:
- IDF version                 ESP-IDF v3.1-43-gd7da7bd-dirty
- Development Env:      [Arduino IDE|Make|Eclipse|other]
- Operating System:     [Windows]
- Power Supply:            [USB]

## Problem Description
我用SDK提供的示例: example/system/ota,编译运行，一直无法成功升级，一直重复操作，日志提示如下：

I (66) boot: ESP-IDF v3.1-43-gd7da7bd-dirty 2nd stage bootloader
I (66) boot: compile time 08:32:26
I (75) qio_mode: Enabling default flash chip QIO
I (75) boot: SPI Speed      : 40MHz
I (81) boot: SPI Mode       : QIO
I (88) boot: SPI Flash Size : 2MB
I (94) boot: Partition Table:
I (99) boot: ## Label            Usage          Type ST Offset   Length
I (110) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (122) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (134) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (145) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (157) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
I (169) boot: End of partition table
I (175) boot: No factory image, trying OTA 0
I (183) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x44198 (278936) map
I (289) esp_image: segment 1: paddr=0x000541b0 vaddr=0x3ffe8000 size=0x00624 (  1572) load
I (290) esp_image: segment 2: paddr=0x000547dc vaddr=0x3ffe8624 size=0x00120 (   288) load
I (301) esp_image: segment 3: paddr=0x00054904 vaddr=0x40100000 size=0x072e8 ( 29416) load
0x40100000: _stext at ??:?

I (324) boot: Loaded app from partition at offset 0x10000

I (6316) ota: Connected to server
I (6316) ota: Connected to http server
I (6326) ota: Send GET request to server succeeded
I (6326) ota: Writing to partition subtype 17 at offset 0x110000
*P*VETTKuU+UZUTT
                TQTQQjPVuZTUQTj*QZUEuP+TQTQ*BBuPTj*QRUEuP+TQTQQBU*uZT(EQVUEj*QVUUUPZBu
                                                                                      U**PUU+VJUZP*+V
uuUZUEUUQUUP
QUUUEUUQVUuJ(Uj*
                UZ)UXUUUTUUT+UZUTUUT
                                    UTUUTUUT*u
                                              UTU+UP
                                                    UUUTUUTUZUTZUUUUJUTUUTBu
                                                                            Q(*RUZUEUUQTQu

T(UP.U]UZJPUVUEUUQTUu
                     -RQQQQQEUTUUTUBu

]Z+P]Q
QQQEQQEUTUUTJU*u
                ,
                 U)Q
QQUQQQUTUUTUu
             ++]E,PQ(QQQQQEQUEUUQTu
                                   ++
E,PQ*Q*QQQUQQUEUUQQu
                    +PKZTTUUTUUTUu
                                  URZUVJV]QUEUUQTU-UVVUZQ
                                                         TQ(QTQTQTUTQU\TQUBXUUUPU-UVVUZ
                                                                                       TQ(QJQQU\QU*Q(jUTUEUUQUU-UVVUZU
                                                                                                                      TQ(QJQUU\QQ(BTUEUUQUUU-UVVUZU
                                                                                                                                                   TQ(QJTQTQQQU*BU*TUEUUQUU*u
                                                                                                                                                                             TJPXUT]UZJPqQQUE

//偶尔出出现如下
Core 0 was running in NMI context:

          00000000 00000004 00000008 0000000c 00000010 00000014 00000018 0000001c 00000020 00000024 00000028 0000002c 00000030 00000034 00000038 0000003c 

3ffecc90  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffeccd0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffecd10  00000000 00000000 00000000 00000000 00000000 00000000 00000000 40100b4a 00000000 00000000 00000000 401034e0 00000000 00000000 00000000 00000000 
3ffecd50  00000000 00000000 00000000 40103127 3ffe9698 00000000 00000000 3ffe9d84 00000002 00000008 3ffe9668 3ffe9d84 00000000 00000000 3ffe9668 4010365d 
3ffecd90  ffffffff 00000020 00000001 3ffe9310 00000010 3ffe9318 00000002 401022b9 00000000 00000000 3ffe9360 401023b4 3ffef054 3ffe898c 3ffe9360 40102780 
3ffecdd0  00000000 00000000 3ffe9318 40101fb4 00000000 00000000 00000000 40102674 00000000 00000000 00000000 3ffef034 3ffe898c 3ffe9314 3ffe9360 401029e0 
3ffece10  3ffe9310 3ffe898c 3ffe9360 00000001 00530022 53002200 00000000 40102ea8 3ffecadc 00000000 3ffecadc 40104bc8 00040000 3ffeed64 3ffecaf4 005c8d2e 
3ffece50  00000000 00000000 2c9f0300 00000040 00000000 00000028 00000000 00000000 00000000 00000000 00000000 00000128 00000017 40104a08 3ffe8600 40105d84 
3ffece90  00000017 40100ad6 00000004 40224b04 40100b23 400044ad 00000023 400044ab 3fff6160 00000000 3fff6160 08000000 60000200 00000000 00000000 00000000 
3ffeced0  0000ffff 00000001 00000002 3ffe8600 3ffee888 00001000 00000128 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffecf10  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffecf50  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffecf90  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffecfd0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffed010  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffed050  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
      PC: 0x40235814       PS: 0x00000033       A0: 0x40104c46       A1: 0x3ffece40 
0x40235814: ieee80211_add_ie_vendor_esp_manufacturer at ??:?

0x40104c46: wDev_ProcessFiq at ??:?

      A2: 0x53002200       A3: 0x00040000       A4: 0x00040000       A5: 0x00040000 
      A6: 0x00000000       A7: 0x00000000       A8: 0x00000000       A9: 0x0000ffff 
     A10: 0x00000001      A11: 0x00000002      A12: 0x3ffe9d8c      A13: 0x53002200 
     A14: 0x00000017      A15: 0x00000128      SAR: 0x00000017 EXCCAUSE: 0x00000000 

我是按照README.md按步骤操作的，请问该如何解决这个问题

xuhongv, 2019-06-10:
- 可以看看这个博文：https://blog.csdn.net/xh870189248/article/details/90079984

wzbhbb, 2019-06-10:
好的，我先看下，有问题再请教，谢谢

北京瑞工科技发展有限公司
王利鹏
地址：北京市海淀区上地东路9号得实大厦 2层中区 5号
邮编：100085 
电话：010 62985019（分机 823）
直拨：010 59487237
传真：010 62983509
手机：18510341619
邮箱：wpeng@rgprt.com
网址：www.rgprt.com 
企业微信名：瑞工打印 微信号：rego-printer
 
发件人： 徐宏
发送时间： 2019-06-10 10:44
收件人： espressif/ESP8266_RTOS_SDK
抄送： wlp; Author
主题： Re: [espressif/ESP8266_RTOS_SDK] examples OTA问题 (GIT8266O-182) (#622)
可以看看这个博文：https://blog.csdn.net/xh870189248/article/details/90079984

wzbhbb, 2019-06-10:
@xuhongv 谢谢您的支持，我现在已经调通了，可以实现OTA升级。在menuconfig中，有俩个地方我补充了下:
Serial flasher config --->
     After flashing (Stay in bootloader)

Bootloader config --->
    [ ] Check APP binary data sum before loading

改动这俩个地方后升级成功率大大提高，不知道是不是和我这里的网速有关，也希望能帮到其它人

xuhongv, 2019-06-11:
第一个修改是无关要紧的，烧录好固件之后不要自动启动程序；
第二个是在 loading 之前不要检验 校验和 ，这个如果不钩上，如果中途下载的不是完整的固件但是又升级成功会出现一些意外情况。

wzbhbb, 2019-06-11:
终于搞定了，感谢 @xuhongv 的提示。我最终是通过git pull，拉取服务器最新的代码，更新后连续测试20多次成功率100%
多谢espressif提供强大的平台

xuhongv, 2019-06-12:
一个乐鑫的合格的程序员，应该时刻盯着sdk版本的更新信息。并且拉取代码同步最新。
@wzbhbb

wzbhbb, 2019-06-12:
就是就是

closed 2019-06-12 by wzbhbb

zhuqiyun, 2019-12-15:
> 就是就是

用的是simple 还是native