espressif/ESP8266_RTOS_SDK #623: Is IDF style ESP8266_RTOS_SDK support RTC memory custom using?

titawork, 2019-06-10:
IDF版本的ESP8266支持像老版本的，可以访问RTC memory 并且自定义内容吗？

dsptech, 2019-06-11:
Hi,
I think you're looking for this:
https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/esp8266/include/esp_attr.h

Regards.

donghengqaz, 2019-06-12:
You can use "RTC_DATA_ATTR". For example:
```
RTC_DATA_ATTR int time;
```

Then when system reboot from deep sleep, the current value of "time" will be keep.

titawork, 2019-06-13:
非常感谢各位大神指导。
但是我尝试过这样申明，结果程序会无法启动，boot会卡在
I (464) boot: Loaded app from partition at offset 0x10000

我的程序只是加了如下代码，所以说，有什么其他的限制吗？
RTC_DATA_ATTR int test1;

void app_main(void)
{
    LOG(_INFO_, "\n******************************************\n");
	LOG(_INFO_, "APP compile time:%s %s.", __DATE__, __TIME__);
	LOG(_INFO_, "\n******************************************\n");
    
    set_log_level(_DEBUG_);
    
    ESP_ERROR_CHECK( nvs_flash_init() );

    LOG(_INFO_, "test %d", test1);
    test1 = 2;
    _mk_InitialiseWifi();
}

titawork, 2019-06-23:
> Hi,
> I think you're looking for this:
> https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/esp8266/include/esp_attr.h
> 
> Regards.

Hello? Did you try to use RTC_DATA_ATTR style values? I tried following, 
It seems do not work. The ESP8266 can't startup But once I delete "test1 = 2;" , the ESP8266 can startup. Do you know what is the problem? Thanks!

RTC_DATA_ATTR int test1;

void app_main(void)
{
set_log_level(_DEBUG_);

ESP_ERROR_CHECK( nvs_flash_init() );

LOG(_INFO_, "test %d", test1);
test1 = 2;

}

dsptech, 2019-06-24:
Hi, it work fine for me.
```cpp
#include <esp_attr.h>
RTC_DATA_ATTR int test1;
char TAG2[]="RTCTest";`

void app_main(void) {
    ESP_ERROR_CHECK( nvs_flash_init() );
    ESP_LOGI(TAG2, "test %d", test1);
    test1=2;
    ESP_LOGI(TAG2, "test %d", test1);
}
```
```
(.....)
I (345) phy_init: phy ver: 1055_12
I (345) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (345) RTCTest: test 0
I (349) RTCTest: test 2
```

In your code I noticed only the missed include, but I don't think that it could be the issue.
Regards.

donghengqaz, 2019-06-27:
Old version may has bug for this feature, so must use new branch like v3.2 and so on.

titawork, 2019-07-01:
Thank you. I will use flash instead for workaround till new version availiable for aliyun SDK.

closed 2019-07-15 by FayeY