espressif/ESP8266_RTOS_SDK #629: socket close() function declaration missing since yesterday

chschnell, 2019-06-14:
After yesterday's upgrades, I believe caused by [modifications to newlib/newlib/include/fcntl.h ](https://github.com/espressif/ESP8266_RTOS_SDK/commit/1b2d3b9cf08306eee28b6195cd4e6cea46ec48d3#diff-2c465153ba6462772b925f673fee24cb), compiler complains about missing function declarations:

```
ESP8266_RTOS_SDK/components/mqtt/ibm-mqtt/MQTTClient-C/src/FreeRTOS/MQTTFreeRTOS.c: In function 'esp_disconnect':
ESP8266_RTOS_SDK/components/mqtt/ibm-mqtt/MQTTClient-C/src/FreeRTOS/MQTTFreeRTOS.c:175:5: error: implicit declaration of function 'close' [-Werror=implicit-function-declaration]
     close(n->my_socket);
     ^

ESP8266_RTOS_SDK/components/smartconfig_ack/smartconfig_ack.c: In function 'sc_ack_send_task':
ESP8266_RTOS_SDK/components/smartconfig_ack/smartconfig_ack.c:172:9: error: implicit declaration of function 'close' [-Werror=implicit-function-declaration]
         close(send_sock);
         ^
```
It looks a lot like a missing 

> #include <unistd.h>

 statement.

SzymonKatra, 2019-06-16:
I have the same problem 
```
ESP8266_RTOS_SDK/components/coap/port/coap_io_socket.c: In function 'coap_new_endpoint':
ESP8266_RTOS_SDK/components/coap/port/coap_io_socket.c:162:5: error: implicit declaration of function 'close' [-Werror=implicit-function-declaration]
     close (sockfd);
     ^
```

chschnell, 2019-06-16:
I want to clarify that the missing declarations occur in the SDK's standard component libraries, not in my code. Actually I also had a missing `fcntl()` declaration which I fixed by adding an `#include <fcntl.h>`.

Also, this is the full list of missing function declarations I was able to observe, all of them using network sockets as the file handles:

* `close()`
* `write()`
* `read()`
* `fcntl()`

Last working commit is [Merge branch 'bugfix/fix_i2s_dma_error' into 'master'](https://github.com/espressif/ESP8266_RTOS_SDK/commit/a227a22a36488325468ca007b234a68c27e10692) from June 12th.

Things start to break with the following commit [feat(lwip): Midify stand I/O control APIs declare](https://github.com/espressif/ESP8266_RTOS_SDK/commit/1b2d3b9cf08306eee28b6195cd4e6cea46ec48d3) from June 13th.

I froze my working repository at the last working commit using

`git reset --hard a227a22a36488325468ca007b234a68c27e10692`

which allows me to build again, but now against a repository frozen on June 12th.

chschnell, 2019-06-18:
Today's commit [ed4bbcf610440ee664cf56d69d6ee5392a02e506](https://github.com/espressif/ESP8266_RTOS_SDK/commit/ed4bbcf610440ee664cf56d69d6ee5392a02e506) fixed it, thanks a lot!

closed 2019-06-18 by chschnell